arendst/Tasmota #3053: Counter used for frequency measurement

kajmaj, 2018-06-24:
Actually, is there any possibility to directly measure frequency - pulses per second - by a sensor connected to any available GPIO of Sonoff ? 
It might be usefull for RPM, anemometer, speed etc.
In my case frequency is within the range between 0 and 80 Hz.

Frogmore42, 2018-06-24:
Not directly, and it's not clear how useful it would be.  80Hz is actually quite fast for the way counters in Tasmota work.  You can setup a counter on a GPIO pin and then send that to NodeRED and with simple math get the frequency over the last period.  

(Counter - OldCounter)/(NowInSeconds - OldTimeInSeconds)
That will give the average number of pulses over the period.

Frogmore42, 2018-06-24:
You can read all about counters in the wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands

kajmaj, 2018-06-24:
I found  "pulse Counter or pulse Timer" there in Counters and do not fully understand what does it mean.
I learned  sofar that, for my purpose, it is enough to have it working as per-period counter.
But I do not know (yet) how to achieve it. :(

Frogmore42, 2018-06-24:
The best way to learn about these things is to try them out and see how they behave.  I am not sure exactly how pulse timer works.  It is the inverse of what you requested (frequency) but I have never tried using it.  I did look at the code months ago and did not find it particularly useful for my needs.  The counter is exactly what it says it is.  It keeps a running total of the pulses it has received.  If there is a power failure, it can go backwards, since it does not continually update the flash (which would wear it out much faster).

You can pick a gpio pin and set it to counter 1
That will cause it to start counting pulses.

kajmaj, 2018-06-24:
I' m going to start experimenting. 
I think that it should be possible to reset counter at the end of each telemetry period by a rule.
Rules are another unknown (to me) area.

ascillato, 2018-06-26:
Hi,

The rule to reset a counter every teleperiod is:

```
rule 1
rule on tele-counter#c1>0 do counter1 0 endon
```

For more information see https://github.com/arendst/Sonoff-Tasmota/wiki/Rules

Thanks.

ascillato2, 2018-06-28:
Great!

Thanks a lot for sharing. :+1: 

Closing this issue.

closed 2018-06-28 by ascillato2

riddik14, 2019-01-31:
Is possible read rpm from 5v 3 wired fan with tasmota? And is possible use pwm to control fan power directly?