arendst/Tasmota #3056: WiFi crashes / reboot on versions > 5.12.0 (no matter which core version)

pReya, 2018-06-24:
Hello everyone,

I've read most issue threads here regarding WiFi issues and reboots, however none of them seem to fit my problem: I'm using a Sonoff Basic, that's running v 5.12.0 (compiled with Esp8266 core v2.4.1). This config is rock solid. No restarts whatsoever.

However, whatever newer version I try upgrading to, it will always show reboot / WiFi crash behavior. I tried 6.0.0a with all Core versions (2.3.0, 2.4.0 and 2.4.1) and they all show the same behavior (unresponsive Web interface, approx. 10x blinking green LED, then it goes dark and starts again shortly thereafter).

Will I have to deal with the fact that I'm stuck on v5.12.0 forever?

Thanks in advance!

ascillato, 2018-06-24:
Hi,

Sometimes there is a problem when upgrading. When this happens it is better to erase all flash and flash again by wire.

Jason2866, 2018-06-25:
Hey @pReya,
i have a NodeMCU which isnt working with a Version >5.13. I dont know why!
I tried MANY things nothing helped.... Maybe the hardware differs from others and is incompatible or just somehow defect.
Is it one device you have problems or more?
Sometimes i have problems with devices i play around -> seems dead.
I erase it via esptool (erase_flash) and flash it again -> back to life :-)

closed 2018-07-10 by ascillato2