arendst/Tasmota #3059: Will a Sonoff Basic (3.3 volts) power a HC-SR501 PIR Motion Sensor?

sparkgap, 2018-06-25:
Does anybody know........
Will a Sonoff Basic (3.3 volts) power a HC-SR501 PIR Motion Sensor?
Some places it indicates 3.3v is sufficient.
Will the Sonoff regulator overheat?
The Motion Sensor will be connected to GPIO14.

Lawrencezarb, 2018-06-25:
MMy HC-SR501 is powered by my sonoff, but you need to feed bypass the voltage regulator on the PIR and feed the 3.3v directly to the board. There are several online instrcutions for this. as below

https://randomnerdtutorials.com/modifying-cheap-pir-motion-sensor-to-work-at-3-3v/

sparkgap, 2018-06-25:
Great.  Thanks

closed 2018-06-25 by sparkgap

pcpaulo, 2018-09-15:
i have followed the instruction to bypass the 3.3v, i am using lowest sensibility but my sonnoff still giving false detection.

Any suggestion?