arendst/Tasmota #3060: Sonoff-Minimal not saving Wifi credentials

Lawrencezarb, 2018-06-25:
I have been using the sonoff-minimal file to upgrade mty versions of Tasmota.

To upgrade an older veriosn I would first upload Sonoff-minimal, which in the past would use my existing config, then upload the newer version of Tasomota, again using my exsisting config.

Recently, after upload sonoff-minimal it would not use my exsiting config, and not connect to wifi. 

I use the 4 button press to enabl Wifi manager to input my wifi credentials then press save, but the unit does not remeber the credential and resorts to default, nit connect to wifi as belwo:

00:00:00 CFG: Use defaults

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_0C38FC, GroupTopic sonoffs) Version 5.14.0-minimal-2_3_0
00:00:00 WARNING This version does not support persistent settings
00:00:00 WIF: Connecting to AP1  in mode 11N as sonoff-6396...
00:00:22 WIF: Connect failed with AP timeout
00:00:22 WIF: Connecting to AP1  in mode 11N as sonoff-6396...
00:00:44 WIF: Connect failed with AP timeout
00:00:44 WIF: WPSConfig active for 3 minutes

Jason2866, 2018-06-25:
Migration Path ok?
https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade

Lawrencezarb, 2018-06-25:
HI Jason2866, This does not really address my issue, where Sonoff-Minimal does not save my Wifi credentials, so when it reboots is does not attempt to connect to my inputed credentials.

arendst, 2018-06-25:
By design Sonoff-minimal does NOT save any setting. It is an intermediate firmware to load the final firmware. It uses the already present setting information.

Lawrencezarb, 2018-06-25:
Hi arenst, Yes that is how I was using it. It used to ue the exsiting config setting but now it does not seem to. My biggest problem is trying to get it to connect to the wifi. When I enable the wifimanager I can input my wifi credentials, but upon rebooting it does not attempt to connect to them.

Frogmore42, 2018-06-25:
Your devices probably have a different cfg_holder than what is in the minimal version.  I believe the default was changed in the last few months.

Lawrencezarb, 2018-06-25:
Hi Frogmore42, that seems resonable enough, but why does the minimal version allow me to save my wifi credentials when in wificonfig mode. Without being able to do this I cannot upgrade to a full version!

SauRoNZA, 2018-06-26:
So let's say I didn't flash the firmware myself originally (as in I have no means to connect via serial) is there any method available to upgrade from Minimal to Full version?

Since I can't get it to connect to Wifi I can't get to the normal web interface. Is there any means to upload the file via WiFi Manager and a sub-URL maybe?

Would WPS work?

reloxx13, 2018-06-26:
just call the upload page and upload:

http://SONOFFIP_WITH_WIFIMANAGER_UP_AND_MINIMAL/up

SauRoNZA, 2018-06-26:
Man if it's that simple I'm going to kick myself! 

I've read the entire internet for a solution but my last resort was logging into the working one I have when I get home and checking what the URL is to try that on the "broken" one.

SauRoNZA, 2018-06-27:
@reloxx13 you are an absolute legend, that did the trick.

Bizarrely my Windows machine wouldn't even see the SSID most of the time and couldn't connect to it. Only my phone would and I managed to upload through that and sort the problem out.

ascillato2, 2018-06-27:
Good to know that your issue is solved. Closing.

closed 2018-06-27 by ascillato2