arendst/Tasmota #3061: Porticsh RF-Bridge-EFM8BB1 upload Error

angustyler, 2018-06-25:
Flashing the RF Bridge with Tasmota 6.0.0a worked perfectly and is now accessible from the web page just fine. However, I'm now trying to install the Porticsh RF-Bridge-EFM8BB1 updated firmware by following the instructions:
[https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433) . But am getting error:
"_Upload Failed. Magic byte is not 0xE9_"
I've using the RF_Bridge.hex found at:
[https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF_Bridge.hex](https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF_Bridge.hex)
(Note: I guess this is the correct location as the folder "tools/fw_efm8bb1" does not appear to exist - unless of course I am looking in the wrong place!)
Any help greatly appreciated!

Jason2866, 2018-06-25:
Try to use the version @arendst includes in folder:
https://github.com/arendst/Sonoff-Tasmota/tree/development/tools/fw_efm8bb1
Do you have compiled the V.6.0.0a with switch (in user_config.h) enabled?

#define USE_RF_FLASH                             // Add support for flashing the EFM8BB1 chip on the Sonoff RF Bridge. C2CK must be connected to GPIO4, C2D to GPIO5 on the PCB

Jason2866, 2018-06-25:
Maybe download (via browser) has corrupted your file...

angustyler, 2018-06-25:
Thank you for your (very quick) response - I think I've moved 2 steps forward and one step back...
I downloaded (zip from Gitbub) and compiled the 6.0.0a source, then installed that using Arduino. 
Then, I uploaded the RF_Bridge_00.hex contained in that zip. This returned the an error stating "Not enough space" (or similar). Since then, when trying to upload, I get "Failed to init RF chip".
From the console, when I call RfRaw with a 0 or 1, I get the expected response. None of the other parameters/payload values seem to get a meaningful result, for example:
16:11:43 CMD: rfraw 255
16:11:43 MQT: stat/sonoffbridge/RESULT = {"RfRaw":"ON"}
I set 
    #define USE_RF_FLASH 1 
before compiling - in case that should be relevant...
Am now stuck for ideas and hoping you may have some suggestions!

Jason2866, 2018-06-25:
Which version of RF Bridge do you have?
R1 or R2 (R1 is in a snap case,  R2 has screws to open)
Erase complete Flash before flashing your compiled Tasmota version!
Do your settings...  For R1 power it over USB and connect C2CK to GPIO4, C2D to GPIO5 on the PCB
Go to Webconsole and upload EFM Firmware. Wait!  The Rf Bridge will beep when done. Success

For R2 power with 3.3V over pinheader (urgent!!!) NOT over USB. Flash will FAIL in this case!!
You can use for powering you USB to serial adapter. Just connect GND and 3.3V.
Connect yes it is swapped!! C2CK to GPIO5, C2D to GPIO4 on the PCB

angustyler, 2018-06-25:
Success! :-) Thank you for the help.
It is an R1 unit. I think the difference may have been making the two GPIO connections and then powering via USB as you suggested. I had been powering it using the 3.3v and ground pins. Maybe that was not delivering enough current to drive the flashing process properly.

Jason2866, 2018-06-25:
Your welcome! Please close this issue.

closed 2018-06-25 by angustyler

Jason2866, 2018-10-02:
Yes you need to connect C2CK to GPIO5 and C2D to GPIO4
to update the EFM chip. If you want you can leave it connected but in this case you have to cut 
Traces for the R2 version.

cslehel, 2018-11-16:
Hello, 
i think i followed the instructions correctly.
Erased the flash with esptool.py, 
connected C2CK to GPIO5 and C2D to GPIO4, 
cut the traces, 
powered with the USB to serial adapter, 
flashed tasmota 6.3, 
when i want to flash the EFM8BB1 with the file from tools/fw_efm8bb1 i am getting the 'Magic byte is not 0xE9' error.

[image]
[image]

Thank you for your time.

Jason2866, 2018-11-16:
Firmware Upload is via web frontend

cslehel, 2018-11-16:
Yes, i am trying to flash the EFM8BB1 firmware at the Firmware Upgrade page Upgrade by file upload section.

Jason2866, 2018-11-16:
And what happens?

cslehel, 2018-11-16:
I am getting Upload Failed Magic byte is not 0xE9 error message when i try to upgrade the EFM8BB1 firmware.

Jason2866, 2018-11-16:
Open efm hex firmware file. Or better upload it here. Maybe corrupt

cslehel, 2018-11-16:
[RF-Bridge-EFM8BB1-20181102.zip](https://github.com/arendst/Sonoff-Tasmota/files/2589797/RF-Bridge-EFM8BB1-20181102.zip)

i copied from /tools/fw_efm8bb1 directory

Jason2866, 2018-11-16:
Do you have selected as Device  Rf Bridge?

Jason2866, 2018-11-16:
Which firmware you are using?

cslehel, 2018-11-16:
On the first page says Sonoff RF Module. I selected on the Configure Module page the 02 Sonoff RF option.  
All GPIO's are 00 None. 
I am using Sonoff-Tasmota 6.3.0

Jason2866, 2018-11-16:
It has to look like this
[image]

Jason2866, 2018-11-16:
You have to select
[image]

cslehel, 2018-11-16:
Thank you very much for your time, i managed to upgrade the efm8bb1 firmware.

liv-in-sky, 2018-12-13:
Hi I'm sorry to open this again - I have the same problem like "cslehel". Same version (6.3) and setting - also same devive R2 v1. I have cut the 2 lines. I got the "Magic byte is not 0xE9" error. 
I'm not sure how to connect the 2 soldered cables.  If I do it like  C2CK to GPIO5 and C2D to GPIO4 - the cables should be crossed ?! The picture above is showing no crossed cables - BUT in any case - I got this error wether crossed or not crossed or with 5V usb or 3.3V . does somebody have an idea?

Jason2866, 2018-12-13:
@liv-in-sky 
You have to select RF Bridge as Modul. 
Error Magic byte is not 0xE9 is a typical indicator that you havent done that

liv-in-sky, 2018-12-13:
I have done all the settings what described above
[image]

liv-in-sky, 2018-12-13:
I have had the 5.12 version on the bridge - I flashed 6.3 over it - could this be a problem ?

Jason2866, 2018-12-13:
Yes, flash this http://thehackbox.org/tasmota/sonoff.bin
And do command: Reset 5
Everything well be reseted and erased. Just Wifi credentials keep intact
After that try again

liv-in-sky, 2018-12-13:
to be honest - I can only flash with atom - could you please tell me which package I have to download for atom - I don't know how to flash only this sonoff.bin file with atom :-(

vigne, 2019-01-11:
Hi All,

same here. I tried all your suggestions, but unfortunately nothing worked. I did 

(1) erase the flash using esptool, 
(2) flashed the file you linked to above (version 6.4.1.8(a124529-sonoff)), 
(3) did a reset 5 in the console (web)
(4) selected "Sonos Bridge (25)" and saved
(5) tried uploading the File "RF-Bridge-EFM8BB1-20181127.hex" and "RF-Bridge-EFM8BB1.hex" using the upload mask i the web front end. 

Unfortunately, neither of the two files worked for me and both aborted with a "Magic byte is not 0xE9" error.

I connected GPIO4 <-> C2D and GPIO5 <-> C2CK, which on my board R2 V1.0 2017.11.23, means the wires are crossed, right? I tried with the switch in "on" position and in "off" position but neither seems to work.

Any help would be very much appreciated!

Jason2866, 2019-01-11:
Have you connected the wire exactly as in the picture?
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433#rf-chip-firmware-upgrade

vigne, 2019-01-11:
Thanks for coming back to me.

I just tried it exactly as shown in the picture (using 5V USB), but no success. But looking on my PCB from the back made me wonder if I might have not "scratched" the PCB deep enough to actually cut the wires. Could you maybe recommend how to do this the safest way so not to damage the rest of the PCB?

And as for the wiring: I also tried powering it with 3.3v from the connector on top (used to flash tasmota), both in the middle (below the switch) and bottom one (next to the speaker). No success really. Also, does the switch have any impact on this? I tried it with both on-off but it failed either way.

Jason2866, 2019-01-11:
Seems everything correct. If you power 3.3v via pinheader cutting traces is not neccessary
You have in main screen the 16 buttons displayed?

h-tro, 2019-01-27:
Hi,

last week I bought a Sonoff RF Bridge (version 433RFBRIDGE R2 V1.0 2017.11.23).
Today I flashed TASMOTA version 6.4 (sonoff DE.bin) successfully.

(1) erase the flash using esptool,
(2) flashed the file sonoff DE.bin version 6.4
(3) selected "Sonos Bridge (25)" and saved
(4) tried uploading the file "RF-Bridge-EFM8BB1-20181127.hex" using the upload mask i the web frontend.

I connected GPIO4 <-> C2CK and GPIO5 <-> C2D
On board R2 V1.0 2017.11.23 the wires are crossed.
The power supply is exclusively via the 3.3V connector.

The same error occurs as described by Vigne:
"Magic byte is not 0xE9"

A new bug?

Jason2866, 2019-01-27:
If you have bridge version 2 you have to be carefull
Dont power over usb. Wont work
Labeling C2CK and C2D is swapped

h-tro, 2019-01-27:
Your answer is a repetition.
I don't use power over usb. 
and the labeling C2CK and C2D is checked several times. The wires are correct.
It seems to be a bug

Jason2866, 2019-01-27:
No, many positive feedbacks. Have you selected Rf Bridge as module?

h-tro, 2019-01-27:
Vigne has already described the same problem.

Yes, Rf Bridge module is selected. That was already written in my first comment.

h-tro, 2019-01-27:
Web-Log
22:39:13 HTP: Hauptmenü
22:39:19 WIF: Prüfe Verbindung...
22:39:19 WIF: verbunden
22:39:20 HTP: Firmware Update
22:39:39 WIF: Prüfe Verbindung...
22:39:39 WIF: verbunden
22:39:45 CFG: in Flash gespeichert am FB, zählen 20, Bytes 3584
22:39:45 UPL: Datei RF-Bridge-EFM8BB1-20181127.hex ...
22:39:46 HTP: Upload abgeschlossen
**22:39:46 Upload: magic byte ist nicht 0xE9**
22:40:00 WIF: Prüfe Verbindung...
22:40:00 WIF: verbunden
22:40:20 WIF: Prüfe Verbindung...
22:40:20 WIF: verbunden

Jason2866, 2019-01-28:
Can you upload you efm hex file?

Jason2866, 2019-01-28:
Sorry i was unclear in what i mean. Could you upload the hex file here in github?

h-tro, 2019-01-28:
Here the hex file.
[RF-Bridge-EFM8BB1-20181127.zip](https://github.com/arendst/Sonoff-Tasmota/files/2804166/RF-Bridge-EFM8BB1-20181127.zip)

Jason2866, 2019-01-28:
Took a look in your hex file. It is garbage!
So error E9 is 100% correct

Jason2866, 2019-01-28:
You are welcome to change the wiki 😀

h-tro, 2019-01-28:
That's not a good idea.
With my english I will spread even more confusion there.

Jason2866, 2019-01-28:
Well this is a argument. Have fun with the RF Bridge

yunnanpl, 2019-03-09:
Dear All, it seems that I need your help.
I have two RF remote controllers for my Sonoff Bridge R2 V1.0, and only 1-2 buttons work on each, so I decided to upgrade the RF firmware. Flashing Tasmota is under control (I did it like 50 times already :)). I had numerous errors while trying to flash RF, but now I am stuck with this "Failed to erase RF chip".
1) I erase ESP and flash clean Tasmota (pre-compiled binary, core 242, from today (but I try to flash RF since 1 week)) (FTDI serial port connected, 3.3v selected, jumper to OFF, with keeping the button, rebooting etc, **this works without problems**).
2) I log onto sonoff Wifi and configure my wifi (it works and connects to the wifi without problems)
3) I switch sonoff off and I connect the extra wires according to the image above
https://github.com/arendst/Sonoff-Tasmota/issues/3061#issuecomment-439417439
4) I configure the Module as Sonoff Bridge (25)
5) I go to Firmware Upgrade, I select the HEX file
6) I get Upload Failed - Failed to erase RF chip

Trouble shooting;
= My Bridge is the R2 V1.0, so I decided to cut the wires according the page below and connect with the USB port (maybe not enough power ?):
https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433
It did not help. So I powered once with USB and once with serial adapter.
= I exchanged the extra wires
C2CK to GPIO4, C2D to GPIO5 and then the other way around C2CK to GPIO5, C2D to GPIO4.
Neither work.
= I flashed different Tasmota versions (core 250, core 242, normal, sensor...). Nothing helps.
= I did the reset 5 trick
= I tried to switch the jumper ON and OFF (not logical, but I was trying everything :))
= I checked the HEX files (redownloaded from different sources, and checked)
I have 5 HEX files (including the original one), not one works.

**One thing which I find confusing, in some posts it is written that the pair button should be pressed during powering. This is probably not correct, as then no WIFI is available (or am I wrong here ?).**

I found on the EFM8 chip description, that one of the connectors should be grounded to enforce booting into boot loader (in case of issues). This did not help either.

As a last resort I wanted to flash EFM8 directly (not through Tasmota), but it seems to be harder than expected (or not possible with serial-USB interface ?).

I will be grateful for any help. This RF thingy is needed for my Home Assistant. Additionally I am planning to integrate door bell (with 433mhz) to log if somebody was ringing to the doors :)

robinleblon, 2019-04-11:
I was facing the exact same problem as @yunnanpl above with R2 V1.0 2017.11.23 board.

What finally did it for me:
- USB power
- Tasmota 6.5.0
- RF-Bridge-EFM8BB1.hex
- properly soldered the wire straps instead of them just having "sit in there"

lobocobra, 2019-09-24:
> Thanks, now the upload works.
> I downloaded the file directly from the website, see screenshot.
> [DL.zip](https://github.com/arendst/Sonoff-Tasmota/files/2804819/DL.zip)
> You should add the hint "Download the whole content via the download button (green) from github" in the wiki.

Can you please update the Wiki, so that others can learn from how you did it? Maybe a picture of your wiring, etc....

Jason2866, 2019-09-24:
Hmm, and what is different to the wiki?