arendst/Tasmota #3062: Sonoff Dual R2: POWER1 ON always after restart (PowerOnState set to 0)

muellererwin, 2018-06-25:
POWER1 of a Sonoff Dual R2 always turns on after restart. Should be off as POWER2 is.
Happens after WiFi Dis/Reconnect (what happens reguarly, 2.3.0?) and after restart.

The parameters PowerOnState, Button/Switch/PowerRetain all on 0.

- [x ] Searched the problem in issues and in the wiki
- [x] Hardware used : Sonoff Dual R2 2017-7-20
- [ x] Provide the output of command ``status 0`` :
```
00:34:09 MQT: stat/sonoff_dual_1/STATUS = {"Status":{"Module":39,"FriendlyName":["Rollo1Rauf","Rollo1Runter"],"Topic":"sonoff_dual_1","ButtonTopic":"sonoff_dual_1","Power":0,"PowerOnState":0,"LedState":1,"SaveData":0,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:34:09 MQT: stat/sonoff_dual_1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:28","StartupUTC":"2018-06-24T23:32:41","Sleep":0,"BootCount":19,"SaveCount":55,"SaveAddress":"F4000"}}
00:34:09 MQT: stat/sonoff_dual_1/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-24T14:30:10","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
00:34:09 MQT: stat/sonoff_dual_1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"...","LogPort":514,"SSId":["...","..."],"TelePeriod":300,"SetOption":["55804009","55C18000"]}}
00:34:10 MQT: stat/sonoff_dual_1/STATUS4 = {"StatusMEM":{"ProgramSize":538,"Free":464,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","17B6179A","00000000"]}}
00:34:10 MQT: stat/sonoff_dual_1/STATUS5 = {"StatusNET":{"Hostname":"sonoff_dual_1-...","IPAddress":"...","Gateway":"...","Subnetmask":"255.255.255.0","DNSServer":"...","Mac":"...","Webserver":2,"WifiConfig":3}}
00:34:10 MQT: stat/sonoff_dual_1/STATUS6 = {"StatusMQT":{"MqttHost":"...","MqttPort":1883,"MqttClientMask":"....","MqttClient":"....","MqttUser":"...","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:34:10 MQT: stat/sonoff_dual_1/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 24 23:34:10 2018","Local":"Mon Jun 25 00:34:10 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"...","Sunset":"..."}}
00:34:10 MQT: stat/sonoff_dual_1/STATUS10 = {"StatusSNS":{"Time":"2018-06-25T00:34:10","Switch1":"ON","Switch2":"ON"}}
00:34:10 MQT: stat/sonoff_dual_1/STATUS11 = {"StatusSTS":{"Time":"2018-06-25T00:34:10","Uptime":"0T00:01:29","Vcc":3.330,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"...","RSSI":100,"APMac":"..."}}}
```

**To Reproduce**
Same behavior with another Dual R2 Tasmota Version 5.14.0 (standard-bin).
This happens when Wifi connection is temporary lost and recovered (maybe due to 2.3.0 with RSS 100%) or the Sonoff is restarted by web config.

**Expected behavior**
POWER1 and POWER2 OFF after restart (the issue is only with POWER1 so far)

**Additional context**
A hint would be great what I could quickly change on the source code as I already compiled 6.0.0a with all the PowerOnState and Power/Button/SwitchRetain OFF. But no improvement so far.

I don't think it's a hardware issue, as it happens for two separated Sonoff Dual R2 on Tasmota plain 5.14.0 and compiled 6.0.0a.

Jason2866, 2018-06-25:
What happens if your mqtt broker is offline and you restart the Dual?

muellererwin, 2018-06-25:
It seems that you are on the right track!
If I delete the MQTT Server form both (5.14.0/6.0.0a) Dual R2s there is no POWER1 ON after restart.
And what would be the solution with MQTT, then?

Jason2866, 2018-06-25:
Your broker has retained messages stored and send this again and again...
I guess you use HASS.io?  Search in issues here there are many...
You have to delete the retained messages. You can search here too. You will find how

muellererwin, 2018-06-25:
Not exactly, but I feel we are close. (Thanks very much for this new solution track)
I use Mosquitto as a MQTT Broker.
I also executed this command to come over any retained messages (at least I thought so):
mosquitto_pub -p 1883 -u <username> -P <password> -d -n -r -t
Any other hint what I could do on the mosquitto side? 

P.S.:
["Why is this also happening with a newly flashed second Dual R2, where nothing happened before? (PowerOnState changed to 0 before compiling)"
Sorry, I know the answer. I used the old MQTT topic.]

Jason2866, 2018-06-25:
Just Mosquitto no HA on top?

muellererwin, 2018-06-25:
It seems it is my time for my outing now. I just very newly started with home automation.  Being proud to handle all this mosquitto / tasmota build / FHEM (good old German HA) / electronics stuff.
I don't even know what HASS.io is, but thought about using IOBroker for integration and easier programming. Is this something similar. Sorry for being a novice.

muellererwin, 2018-06-25:
Thanks for this comment, again. I think I will make my way through forums and google now (as always ;-)
Definitely, no Tasmota issue anymore.

closed 2018-06-25 by muellererwin