arendst/Tasmota #3064: Sonoff Touch Switch

mthiaku, 2018-06-25:
Not able to get FinalStage SSID during over air flushing to tasmota.

Jason2866, 2018-06-25:
Since we are not an clairvoyant we need infos....

mthiaku, 2018-06-25:
I’m learning to flush tosmota over TI EU 2 Gangs Smart wifi switch ITEAD_1000xxxxxx and on 2nd phase not able to complete. Using windows platform with python.exe sonota.py to connect WiFi and uses pairing mode to load. No error. Missing FinalStage SSID to reboot.

mthiaku, 2018-06-25:
Do I need to invest over FTDI?

Jason2866, 2018-06-25:
To be honest i never used the OTA method because i have sonoff basics with new firmware...
So i always flash via serial interface.
To answer your question i think yes. It is much easier to flash and it ALWAYS works.
Dont buy a FTDI device. The arent good here. The deliver not enough current on the 3.3V rail.
Buy a cheap USB serial Adapter with a CH340 chip which can switch between 5V and 3.3V!

Snipercaine, 2018-06-25:
if you get to the "Final Stage" but no further, that's the sign that your sonoff is running the notorious v1.6 firmware that disabled our OtA flash-ability.  There is supposed to be a newer version of the firmware shipping now that allows OtA flashing again, but I haven't seen any in the wild. 
I would totally recommend getting an FTDI adapter and a few jumper wires and just do it manually. It's not hard once you get into it.  And it's more reliable I think.  I haven't done OtA in many months now.

Geitde, 2018-06-27:
FTDI adapters fail as they drain to much power from usb. Usually the result is that the usb device disconnects and reconnects all the time. It may work for a desktop machine, but usually laptops have even more weak usb power and are unable to power the FTDI adapter and the sonoff device connected to it.

However. You can just connect GND, TXD and RXD and NOT connect the VCC line. Now you can power up the sonoff using mains power, but **be aware that this is quite dangerous** and a little mishandling may destroy your laptop or cause harm. You should know what you are doing while dealing with mains. 

This method is totally fine, when you know what you are doing. I flashed all my sonoffs using FTDI and mains power.  Since my devices are mounted inside the walls I have not much choice to flash as I cannot power down everything just to flash stuff.

I always wonder why there is no real boot loader. e.g. in the first 128KB of memory allowing to flash the entire device (beside that 128KB boot loader) OTA and in one step. e.g. using TFTP and some default server. This minimal.bin method is just failing somewhere in the future and the bootloader net and flash code could be reused by the real firmware, so no big space lost.

Frogmore42, 2018-06-30:
The esp8266 has a built-in bootloader in ROM, so it can never be accidentally erased/changed.  But it is small and has no networking.

If you are asking why isn't there a secondary bootloader, the probable answer is it would need all the SDK Wi-Fi code plus other stuff.  All that would need to be duplicated in the main code or it would difficult to update it for the main code.  Since it is Wi-Fi code in the secondary bootloader, it too would need to be updatable.  The linking process would need to be more complex too.

It can be done.  Take a look at Particle.io That is how they do it.  The Arduinoish code users write gets compiled and flashed by the underlying firmware.  The underlying firmware can also be updated OTA.  Of course they have been doing it awhile.  

You are right that at some point, the code might grow too much to fit even with minimal.  At that point, there are a few choices to make one or both pieces smaller.

mthiaku, 2018-07-03:
Perhaps it EWeLink version not the sonoff. Any idea how to check firmware version.

Snipercaine, 2018-07-03:
to see the version of the firmware on the Sonoff device you have to first connect it via the ewelink app, then there's a "device settings" button and that's where you'll see the firmware version.  
I believe (but haven't tested myself), that if you have the dreaded version 1.6, you can upgrade the stock firmware to v2.0 and then you'll be able to flash with SonOtA.exe 

But if you have an FTDI in the mail, are you still going to OtA flash?

mthiaku, 2018-07-04:
It was purchased first week of June and comes with version 2.0.1
[image]

mthiaku, 2018-07-05:
Received FTDI and manage to solder the jumpers. 
Do I need to short circuit 5V and VCC and use 3V3?

[image]

Geitde, 2018-07-05:
I have the same version of the device and had the same question. I simply removed (and lost) the jumper. That way it always deliveres 3.3V on the lines and I never have to question where or if it needs to be set when it falls off.

ascillato2, 2018-07-10:
Hi @mthiaku 

Have you manage to solve your issue?

ascillato2, 2018-07-17:
Hi @mthiaku

If you still have this problem, please, reopen this issue and give us more information so we can help you. Thanks.

closed 2018-07-17 by ascillato2