arendst/Tasmota #3067: sonoff basic always hangs after watchdog reset

hackbar, 2018-06-26:
Whenever the device resets due to the watchdog resetting, it hangs indefinitely. It's a sonoff basic module.

Is there anyway to make it reset when the watchdog fires?

Log (it is formatted like this, with the extra newlines)
00:00:54 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(1,1)

 ets Jan  8 2013,rst cause:4, boot mode:(1,1)

wdt reset

Frogmore42, 2018-06-26:
The whole point of a watchdog is to actually reset the device.  If it is not working, it is probably a device specific hardware issue, ie your Sonoff is broken due to one or more of many reasons.  It's probably time to get a new one.

ascillato2, 2018-06-29:
Hi,

If your question is addressed, please close this issue. Thanks :+1:

closed 2018-07-10 by ascillato2