arendst/Tasmota #3068: How to cancel setting RF switch in SONOFFT1

SUGINACH, 2018-06-26:
I Flash MY SONOFF T1
And set a RF switch to turn on the device
Such https://www.aliexpress.com/item/Newest-SONOFF-T433-86-Type-Wall-Panel-433MHz-Wireless-RF-Remote-Control-Transmitter-Support-All-SONOFF/32852333652.html
But I do not know how to undo the setting
I want the device not activated by this switch anymore
I tried to connect the T1 to another RF switch but it did not cancel the previous RF Switch

arendst, 2018-06-26:
See https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF#procedure

SUGINACH, 2018-06-26:
waht about sonoff t1?

arendst, 2018-06-26:
Same procedure. Try it first and come back with your result.

SUGINACH, 2018-06-26:
I probably do not understand
I have a SONOFF T1 with two channels
One of the channels has been set up for a remote control that I do not want to control this channel
If I press this channel's touch button three times nothing happens it just turns on and off and turns on
I apparently did not understand what I was supposed to press

SUGINACH, 2018-06-26:
PS
Your firmware is the best thing that happened to me in the context of a smart home
Thanks for everything
Keep up with the good work
I hope it will benefit you at least as it is beneficial to me

arendst, 2018-06-27:
On the PCA of the T1 is a separate small button to reset both 434MHz remembered signals.

SUGINACH, 2018-06-27:
In the end, I got along differently
I pressed the touch button until two beeps sounded
Then I pressed the RF I wanted to delete
And the connection between the remote control and the button has been canceled
thanks anyway
And again
Thank you very much for all the investment in this wonderful software And on support

ascillato2, 2018-06-27:
Great to know that your issue is solved. Closing.

closed 2018-06-27 by ascillato2