arendst/Tasmota #3069: Sonoff relay work with 3,3V but no with 220V

MarcioMarques86, 2018-06-26:
i already did flash with my own firmware for MQTT to Home assistant, and the Sonoff connects to the wifi, and relay works well when i have 3,3v connected to the sonoff, but when i connected just 220v, the relay doesn´t work. Anyone knows what is happened? thank you﻿

Frogmore42, 2018-06-26:
The simple answer is, with the limited information you provided, no.
There are many things than can be wrong.  You need to provide all the information from the template if you expect anybody to be able to help you.

But here is a hint.  If the relay worked when you had it connected to 3V3, it wasn't really 3V3.  Typically Sonoff devices have 5V relays and they don't work when connected to only 3V3.  It is possible that your device has a 3V3 relay.  It is also possible that your device has been damaged by being connected to 5V.

There are also other possibilities.

MarcioMarques86, 2018-06-26:
Thank you Frogmore42. my model of Sonoff is: Sonoff Basic. And i have 2 of these, and the behaviour of these two is the samething. Sorry, i think that i didn´t explain well, i flashed my Sonoff with FTDI and your power source (3,3V) and the relay doens´t work, because as you say, the coil need 5V. For relay work well i had connect to 5V. But then i disconnect the 5V and connect 220V and the relay doesnt work. but if i try again with the 5V, the relay works well again.

andrethomas, 2018-06-26:
In my experience the relay on the sonoff basic still clicks when connected with a TTL usb connector. The fact that it does not work off 220V suggests there is either something wrong with the on board power supply module which brings the 220V down to 5V, or the 3.3V regulator which brings the 5V down to 3.3v.

Do some voltage measurements on the pins when connected to 220V and see if you get 5V on the power supply module first. If there's 5v there then the 3.3V regulator most likely faulty.. if no 5V then power supply faulty.

[image]

**Be very careful when connected to 220V as there are live traces that can harm or kill you.**

MarcioMarques86, 2018-06-26:
Guys, thank you, but i changed the Input with the Output! that's a great distraction..ufff.

ascillato, 2018-06-26:
Ufff

Great that now it is working for you.

Please, close this issue. Thanks :+1:

closed 2018-06-26 by MarcioMarques86