arendst/Tasmota #3070: Sonoff POW R2 Tasmota, Cellar sump pump

c3reeman, 2018-06-26:
Hi woundering what my option are and if this is possible, I’ve fitted a sump pump into my cellar / basement and I want to monitor how often it runs... I figured a good way to do this would be to use a pow r2 ?? and I’d be able to see the amps spike when the pump runs, I’m not sure if home assistant will have enough resolution to capture each run though as when it does run it’s not for long at all, probebly about a minute does anyone have any suggestions ?

Frogmore42, 2018-06-26:
Yes, you can do this.  I don't use Home Assistant.  With the number of people that have problems with it, I haven't been interested in investing the time to learn it.  I use NodeRED to control & monitor my dehumidifier using a POW to measure power and a wemos D1 to measure temperature and humidity.  I also have a modified tipping bucket rain gauge I use to measure how much water is generated.  I have the POW set up to send results every 10s and flows in NodeRED that read that plus the various sensors to make a determination of when to turn on the dehumidifier and when to turn it off.  With my control, the efficiency water/KWh is significantly greater and the average humidity is reasonably well controlled and the temperature is lower (less wasted heat).

ascillato2, 2018-06-29:
Hi,

If your question is addressed, please close this issue. Thanks :+1: