arendst/Tasmota #3071: AIlight receives mqtt  OFF command  from  unknown  source - so can't  hold its state

gnkarn, 2018-06-26:
ailight , after turning on , receives an off command from unknown source , so it can't hold its state.

here the logs after turning it on

12:07:12 RSL: Topic Recibido cmnd/RGBliving1/power, Tamaño de Datos 2, Datos ON
12:07:12 RSL: Grupo 0, Índice 1, Comando POWER, Datos ON
12:07:12 MQT: stat/RGBliving1/RESULT = {"POWER":"ON"}
12:07:12 MQT: stat/RGBliving1/POWER = ON
12:07:13 CFG: Grabado a la flash en F7, Conteo 1733, Bytes 3584
**12:07:15 APP: Botón1 multi-press 1
12:07:16 SRC: Button
12:07:16 MQT: stat/RGBliving1/RESULT = {"POWER":"OFF"}
12:07:16 MQT: stat/RGBliving1/POWER = OFF**
12:07:16 CFG: Grabado a la flash en F6, Conteo 1734, Bytes 3584
12:07:18 HTP: Comando
12:07:18 SRC: WebCommand from 192.168.0.61
12:07:18 RSL: Topic Recibido /Status, Tamaño de Datos 1, Datos 0
12:07:18 RSL: Grupo 0, Índice 1, Comando STATUS, Datos 0
12:07:18 MQT: stat/RGBliving1/STATUS = {"Status":{"Module":27,"FriendlyName":["luz_living1"],"Topic":"RGBliving1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:07:18 MQT: stat/RGBliving1/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T02:22:19","StartupUTC":"2018-06-26T12:44:59","Sleep":0,"BootCount":11,"SaveCount":1734,"SaveAddress":"F6000"}}
12:07:18 MQT: stat/RGBliving1/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-23T18:49:39","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:07:18 MQT: stat/RGBliving1/STATUS3 = {"StatusLOG":{"SerialLog":1,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WIFIgusPb",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
12:07:18 MQT: stat/RGBliving1/STATUS4 = {"StatusMEM":{"ProgramSize":508,"Free":492,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00002C0A","0F4AA794","0C000000","02000092","00000000"]}}
12:07:18 MQT: stat/RGBliving1/STATUS5 = {"StatusNET":{"Hostname":"RGBliving1-6346","IPAddress":"192.168.0.26","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"DC:4F:22:0C:B8:CA","Webserver":2,"WifiConfig":1}}
12:07:18 MQT: stat/RGBliving1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.60","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0CB8CA","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:07:18 MQT: stat/RGBliving1/STATUS7 = {"StatusTIM":{"UTC":"Tue Jun 26 15:07:18 2018","Local":"Tue Jun 26 12:07:18 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"00:48","Sunset":"16:58"}}

> 
> Tested with QOS 1 and 0 , no difference .
> i tried to find a way to disable any button action ( even when nothing has been configured , and no button is defined for an Alight) , but couldn't find any place to do it .
> 

### **this is my config on the Home assistant side** 

--------------

- platform: mqtt
  name: "luz living1"
  command_topic: "cmnd/RGBliving1/power"
  state_topic: "stat/RGBliving1/RESULT"
  state_value_template: "{{ value_json.POWER }}"
  rgb: true
  on_command_type: "last"
  rgb_state_topic: "stat/RGBliving1/RESULT"
  rgb_command_topic: "cmnd/RGBliving1/color"
  rgb_command_template: "{{ '%02x%02x%02x00' | format(red, green, blue) }}"
  brightness_state_topic: "stat/RGBliving1/RESULT"
  brightness_value_template: "{{ value_json.Dimmer }}"
  brightness_command_topic: "cmnd/RGBliving1/Dimmer"
  brightness_scale: 100
  white_value: true
  white_value_state_topic: "stat/RGBliving1/RESULT"  
  white_value_scale: 100
  white_value_template: "{{ value_json.Channel[3] }}"
  white_value_command_topic: "cmnd/RGBliving1/channel4" # formato de 0-100
  availability_topic : "tele/RGBliving1/LWT"
  payload_available: "Online"
  payload_not_available: "Offline"
  qos: 0
  payload_on: "ON"
  payload_off: "OFF"
  retain: true

Jason2866, 2018-06-26:
Your broker has stored retained messages...
Many Hass users have problems with that.
Delete the retained messages (search in issues here for solution)
Change you config in general not using function "retain". It makes more problems than it solves.
I cant help you with Hass, i use a other HA system (OpenHAB).

gnkarn, 2018-06-26:
thanks for your answer , i changed it , restarted the broker , and made no difference .

RSL: Topic Recibido cmnd/RGBliving1/power, Tamaño de Datos 2, Datos ON
14:11:40 RSL: Grupo 0, Índice 1, Comando POWER, Datos ON
14:11:40 MQT: stat/RGBliving1/RESULT = {"POWER":"ON"}
14:11:40 MQT: stat/RGBliving1/POWER = ON
14:11:41 CFG: Grabado a la flash en F7, Conteo 1765, Bytes 3584
14:11:41 WIF: Probando conección...
14:11:41 WIF: Conectado
14:12:01 WIF: Probando conección...
14:12:01 WIF: Conectado
14:12:08 SRC: MQTT
14:12:08 RSL: Topic Recibido cmnd/RGBliving1/channel4, Tamaño de Datos 2, Datos 23
14:12:08 RSL: Grupo 0, Índice 4, Comando CHANNEL, Datos 23
14:12:08 MQT: stat/RGBliving1/RESULT = {"POWER":"ON","Dimmer":66,"Color":"A8A8003A","HSBColor":"60,100,66","Channel":[66,66,0,23]}
14:12:08 SRC: MQTT
14:12:08 RSL: Topic Recibido cmnd/RGBliving1/power, Tamaño de Datos 2, Datos ON
14:12:08 RSL: Grupo 0, Índice 1, Comando POWER, Datos ON
14:12:08 MQT: stat/RGBliving1/RESULT = {"POWER":"ON"}
14:12:08 MQT: stat/RGBliving1/POWER = ON
14:12:08 CFG: Grabado a la flash en F6, Conteo 1766, Bytes 3584
14:12:16 APP: Botón1 multi-press 1
14:12:16 SRC: Button
14:12:16 MQT: stat/RGBliving1/RESULT = {"POWER":"OFF"}
14:12:16 MQT: stat/RGBliving1/POWER = OFF
14:12:21 WIF: Probando conección...
14:12:21 WIF: Conectado

what is strange to me , are this lines
14:12:16 APP: Botón1 multi-press 1
14:12:16 SRC: Button

what do they mean? ( APP, SRC ????)
because the problem appears after them as you see on the logs.

the change to the "retain" parameter on the message added a new problem, now the "White" change do not work at all, it remains at 100% and cant even be changed  on the home assistant slider , but this probably is a home assistant issue , i will get rid of the home assistant light component and send the different messages individually - 

it is a pity , this AIlight configuration ,  do have some problems , is the only one i couldnt make to work properly .

ascillato, 2018-06-26:
APP means from the tasmota application
SRC means Source of command.

So, this means that you have a button pressed.

gnkarn, 2018-06-26:
no it is not the case ,

there is NO button on this device ( Alight) , the button does not exist on the configuration and of course not physically on the device , that is why i suspect from a bug .

Jason2866, 2018-06-26:
Just to be sure. Shut down the broker (power off) and look what the lamp does.

ascillato, 2018-06-26:
Hi,

In the configuration it has a button:

```
  { "AiLight",         // Ai-Thinker RGBW led (ESP8266 - my9291)
     GPIO_KEY1,        // GPIO00 Pad
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor pad
     GPIO_USER,        // GPIO02 Optional sensor SDA pad
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor pad
     0, 0,
     0, 0, 0,          // Flash connection
     0, 0,
     0,                // Flash connection
     0,
     GPIO_DI,          // GPIO13 my9291 DI
     0,
     GPIO_DCKI,        // GPIO15 my9291 DCKI
     0, 0
  },
```

I don't have that device to test, but if it does not have physically a button, it should be taken out from the configuration.

Please, try to modify the _sonoff_template.h_ line 600 from `GPIO_KEY1` to `GPIO_USER`

gnkarn, 2018-06-26:
ok, this is what i think happened , because i was still working on the lightbulb, i left the  wire (to be able to reset the ESP , soldered in ) , and i think the this was enough to pick random induced noise ( anthenae ) , and this was causing wired behavior .. will see , tks

ascillato2, 2018-06-29:
Hi,

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-29 by gnkarn