arendst/Tasmota #3073: Sonoff External Push Button Switch Setup and Sonoff Remote Access without WiFi

mxplea10, 2018-06-26:
I'm using Sonoff Basic, Dual, 4CH Pro, Pow and TH16 switches in two secured network cabinets.  Each Sonoff switch is installed in a flame retardart enclosure and a ladder is required to access both cabinets.  

1.  I'd like to install/wire an external push button switch that duplicates the on-board switch functions of each of the specific Sonoff devices.  Otherwise, the enclosure will have to be opened each time the button function is required.  I haven't been able to find documentation for installing an external push button switch for specific Sonoff switches.

2.  I'd also like to be able to access the Sonoffs remotely whenever WiFi isn't available.  Can this be done using bluetooth with an iPhone or with another type of device?  The only documentation that I've located is ["Control a Sonoff using a remote button"](url) but it seems to be incomplete.  A schematic is mentioned in this article but it isn't available and there's no details and/or configuration on how to setup the Power Regulator, DHT and ESP01 to communicate with the Sonoff.  Is there additional information available for this or is there a better way to do this now?

Can someone provide additional information or recommend how I can accomplish installing an external push button for each type of Sonoff and access the Sonoff remotely without WiFi?

Jason2866, 2018-06-26:
To connect a switch/button look here
https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs

No wifi no remote controlling.

ascillato, 2018-06-26:
A sonoff device only has WIFI, no bluetooth.

mxplea10, 2018-06-26:
Please note that I stated that I want to setup an external push button switch that duplicates the on-board button functions ([](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage)).  Looking at  https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs[](url) ("* GPIO_KEY1,           // Button usually connected to GPIO0)", are both wires of a two-way push button switch connected to GPIO0? 

The remote control article that I referenced above is here [](https://github.com/arendst/Sonoff-Tasmota/wiki/Control-a-Sonoff-using-a-remote-button).  A Sonoff combination WiFi - RF device supported by Tasmota might be what I'm looking for.  I'm not sure if this, Sonoff RF- WiFi Wireless Smart Switch With RF Receiver []((https://www.itead.cc/sonoff-rf.html)) is the type of device that I can use for both WiFi and RF access.

ascillato, 2018-06-26:
The Sonoff RF (https://www.itead.cc/smart-home/sonoff-rf.html) is a RF 433Mhz AND Wifi device

ascillato2, 2018-06-29:
Hi,

If your question is addressed, please close this issue. Thanks :+1:

Frogmore42, 2018-06-30:
For GPIO 0, one wire goes to ground, the other to the GPIO.  Make sure you use a momentary switch/button.

closed 2018-07-02 by mxplea10