arendst/Tasmota #3075: Router Auto-Rebooter

adamrhoades, 2018-06-26:
I was wondering if it would be possible to include an option for monitoring an internet connection and restarting a modem by turning the Sonoff off and on again in the event connectivity is lost for several mins?

I've seen a few products like this, and a couple of hombrew Sonoff solutions, but would love to be able to do this via Tasmota

Example : https://captain-slow.dk/2016/07/09/automated-router-restarting/

Frogmore42, 2018-06-26:
I have seen a few people ask similar questions and I have to ask why?  Is your modem/router really that unreliable that a simple reboot fixes it?  If so, perhaps it is time to get a more reliable one.  A few years ago, I had a cheap one at the barnyard.  Every few months it would loose its mind and stop working.  I could have spent a lot of time and effort putting in place a system to reboot it when the problem occurred.  It might have even been fun to do that.  But, after I finally got tired of manually rebooting the router,  I replaced it with an equally cheap (but much more reliable) one.  The new one never needs a reboot.  It does get one every few months (for the 30s it takes the back up generator to determine the power really is out and is not likely to come back very soon).

The modem and main router are on a UPS so they never lose power.  While my internet has gone out many times in the last several years, not once would rebooting the modem or router have restored service (since the problem was really at the provider).

I am also not sure what benefit Tasmota could bring to this space, ie why do you want this capability via Tasmota?  As you said there are already solutions to this problem.  What is wrong with them that you believe Tasmota would do better?

Jason2866, 2018-06-27:
@Frogmore42  agree with you 100%. Go get a reliable router....
Same here. NONE of my -few- problems with internet connection the last years could have been solved with simply rebooting the router (cable cut, power supply fail, switch died, overvoltage damage)

adamrhoades, 2018-06-27:
I'm afraid I have to disagree with you both. Here in the UK, if you have a cable DSL service with Virgin Media, you have no choice but to use their hardware. People like me who have become dissatisfied with this equipment have replaced all the routing hardware, access points, controllers etc with better equipment. I myself have invested heavily in unifi by ubiquity. I am still forced to use Virgin Media's router in what they call 'Modem Mode'.  All of this hardware is protected by a UPS as you suggested.

Its usually very robust, but recently there have been whole area outages and twice the 'Modem' has not reconnected on its own when the service came back up. The most recent of which happened whilst I was on holiday. I lost all connectivity to my CCTV and smart home monitoring and was convinced I had been burgled in my absence.

My only recourse was to contact friends and have them drive to my house, let themselves in and restart my modem.

As for why would I like to use Tasmota? The other solutions are essentially dumb, self contained autonomous devices once configured. Any number of problems could result in the router being restarted.  I would have no ability to remotely monitor, log, or change parameters to troubleshoot on the fly.

My eventual plan is to add a 4G(Cellular) failover to my gateway hardware for redundancy, which would allow me much more control in the event of an outage.

Jason2866, 2018-06-27:
That is a reason. Hating it, to use dongled (crappy and very often unsecure) hardware.
In germany this was not long ago the same. But now cable providers has to allow user owned hardware.
For DSL many many years ago, the same thing.

arendst, 2018-06-27:
@adamrhoades In my case I have a few routers connected to my ADSL access router (would be your DSL router). These routers service tasmota devices. I also run my own DNS (dnsmasq) and NTP server on a NAS providing Domoticz Home Automation (HA) too. This way, losing the ADSL connection everything at home keeps un spinning.

You might want to run a small node-red service on a your HA server pinging the internet every minute. When no reply is received let it toggle a tasmota driven device controlling your DSL router.

Frogmore42, 2018-06-27:
Okay, there are some providers that haven't learned how to make things reliable, and there are people who are stuck using equipment from them.  I am still not clear what Tasmota would bring to the this solution.  What are your requirements for this solution?

Watchdog development is hard.  Making a mediocre one is easy.  Making one that always does what you want/need and never what you don't much, much, much harder.

As an example, a few nights ago I was working on my laptop and stuff that had been working stopped working.  I looked down in the sys tray and saw that Windows had detected a problem with the Internet.  Sure enough, I was not able to see the outside world.  This had happened a couple of days earlier, but at that time the power was also out.  This time the power was on.  I went to the power provider's web site to see if there were any outages, but there were none.  I then tried to go to the internet service provider's web site, but it was really slow.  I finally got there, but they had no information.  I called them and finally got a recording about the outage.  Since it was almost midnight, I figured they must have scheduled a more permanent repair from the previous outage.  In neither of these cases would it have been a good thing to have rebooted the modem or router.  Now, it probably would have hurt nothing to reboot just the router, since it had no service.  There was a warning to not reboot your router at a previous outage, but I expect that would be really rare.

If you want something configurable, you can use NodeRED or perhaps some other HA software to get exactly the behavior you want.  This would be something you could implement today.  It would also help you understand exactly what behavior you need.

adamrhoades, 2018-06-27:
Useful feedback, thank you. I had considered automating this with my HA system, but I've been using Samsung Smartthings which is pretty much useless if you loose connectivity.

I'll have a look at NodeRed for this task and see what I can come up with.

ascillato2, 2018-06-28:
Hi,

If your question was addressed, please close this issue. Thanks :+1:

andrethomas, 2018-06-29:
I would run node-red on a pi right next to the router connected with physical cable... when node red cannot ping external ip address like google you can use one of the GPIO's to remove power from the router using a relay breakout board for a few seconds and turn it back on. 

So basically as soon as node-red does not get ping returns from a remote host it will turn power (the 9v or 12v cable through a relay) off for a few seconds and then on again.

closed 2018-06-29 by adamrhoades

Frogmore42, 2019-01-12:
Perhaps read a little closer.  The request was for requirements for the requested feature.  I don't need this feature.  My router has recently started loosing WiFi connectivity, but I am not sure I really want to have a Tasmota device sitting there rebooting the router every time it can't see something.  I have written a watchdog (several actually).  It is really hard to write a really good one.

So, what does a good one look like to you?
What configuration capabilities does it need to be generally useful for different scenarios?

There is probably something that can be done with rules.   Maybe even with rules as they exist today.

slowtraveler, 2019-04-16:
I've solved the problem with Sonoff S20 with Tasmota firmware... My router sometimes just freeze, and always loose WIFI at that time... Just do the following:
1. Set WifiConfig to 5 (disable Wi-Fi config but retry same AP without restart and flash writes)...
2. Set PowerOnState to 1 (output ON upon powering on the device, for faster booting the rooter)
3. Activate two simple rules:
first - triggering on Wifi#Disconnected: switch relay off, delay 30 sec, switch relay on (also added delay 3 minutes after this, but i'm not sure if it's required). Backlog command is required for executing sequence of commands within the rule/trigger 
second - trigerring on Time#Minute|5 , switch relay ON ( every 5 or 10 minutes switch relay on even if no connection or status is OFF).

Result - upon loosing Wifi, switch will stop the output power to the router, and will restart it after 30 seconds. Meanwhile will try to reconnect to the same AP (router's).
If you remotely switch off the switch, you will loose the remote connection to it. But it will power itself on within 5 or 10 minutes.

NorthernMan54, 2019-05-09:
@slowtraveler Would it make sense to add this to the "Rule Cookbook" on the wiki ?

https://github.com/arendst/Sonoff-Tasmota/wiki/Rule-Cookbook

slowtraveler, 2019-05-12:
I am not sure if the solution/code is 'good enough' for Cookbook...
Main reason - it generates SWITCH ON command each several minutes, which is reported to MQTT broker and 'floods' log. Unfortunately at the moment i cant avoid this - 'cannot put IF command into the rule to check if status is changed and to report to the MQTT broker only upon changing the status'...
Anyway - just to simplify explanation: My solution was:
`rule1 on wifi#disconnected do backlog power 0;delay 150;power 1; delay 1800 endon
rule2 on Time#Minute|10 do power 1 endon
WifiConfig 5`
So - rule1 : upon disconnect, switch router off, wait 15 seconds (think it's enough for 'cold boot'), power router on, and wait 180 seconds for router to boot (if no waiting - sonoff will reboot router in loop, because router cannot start WIFI immediately.
rule 2: just power router ON each 10 minutes. That's in case someone (or you) incidentally switch off router remotely -  you cannot issue the command for powering it on back.
Wificonfig 5 - don't try to access other AP, but just wait and connect to AP in interest... Otherwise Sonoff can connect to other AP and 'wait' even if router is not working...
Trust above will work for people. It's working for me several weeks already (and i rebooted the router several times meanwhile). If someone can solve the problem with POWER ON MQTT reports - kindly advice...
Also if you think it can be added to Rule CookBook - please feel free to post it there.

NorthernMan54, 2019-05-15:
I'm just testing this now, I needed to flash up a couple of new devices.  I think that if is a working pattern, publish it, with the logging caveat.  Worst case is, someone improves it.

slowtraveler, 2019-05-15:
OK - PLS revert feedback for testing :)
Meanwhile try just following rule:

> rule1 on wifi#disconnected do backlog power 0;delay 150;power 1; delay 1800; power 1 endon

rule2 is not needed if poweronstate is 1 (power on the load upon booting). If you intentionally power the router off, it will be powered on back upon executing of RULE1 (you will loose wifi at powering router off).
The reason for added second 'power 1': when testing, i revealed that when device is not connected to server upon 'powering on', logically server is not received POWER ON state (even upon connecting). So it continues to report that router is POWERED OFF even after rebooting and despite the fact that relay is already ON. So after waiting to connect, device is issued once more time POWER ON state, just to inform server...

NorthernMan54, 2019-05-15:
Tks for the rule 2 tip

In my setup, I’m also going to use node-red to execute a ping test and turn off the router if it fails.  I would presume most people would have node-red as part of their setup.

slowtraveler, 2019-05-15:
You cannot send command to the RELAY when router fails :) 
I'm not using node-red :)

NorthernMan54, 2019-05-16:
The use cause I’m dealing with is wifi working, but connection to internet is not working and reseting the modem is required to reconnect.

This is a node-red flow, that pings google every 5 minutes, and if the ping fails, send a turn off message to the router.  To prevent a loop constantly turning off the router, it sends the message once, and doesn’t try to turn off again until 10 minutes after the last ping failure ( recovered for more than 10 minutes ).

Am thinking I also need a watchdog as well, so that if the tasmota rule turns back on the router, but it doesn’t reconnect to the internet, reset the router again.

Maybe I need to think about this flow further.

slowtraveler, 2019-05-16:
For internet connectivity i'm using router's internal watchdogs.
And in fact if router start to reboot due to watchdog, tasmota is powering it off and making cold boot due to above rule :)
In case router is not started upon reboot, tasmota reboots it again after 1800 seconds - that's why i put in rule second delay. Otherwise tasmota reboots router even before router can boot and power on it's WIFI :) You can try it when 'relay' is not connected to router - just power off the router and look how relay will cycle after expiring the delay period...

Cr4z33, 2020-04-28:
Forgive me if this is a stupid question, but can this work also if the Wi-Fi network stopped working? 🤔

I need to replace my AP because it really sucks, but in the meantime I was thinking about this solution (useless if not working without WiFi service on 24/7).

slowtraveler, 2020-04-28:
> can this work also if the Wi-Fi network stopped working? 🤔

In fact it works ONLY if  Wi-Fi stop working :) Because triggering logic is:
If you cannot find WIFI (you are disconnected) than 
1. Power off the Relay (where 'hanging' router or AP is connected)
2. Wait few seconds (to discharge capacitors in router/AP for correct powering on).
3. Power relay back on

Cr4z33, 2020-04-29:
> 
> 
> > can this work also if the Wi-Fi network stopped working? 🤔
> 
> In fact it works ONLY if Wi-Fi stop working

OK thanks I just ordered 4 more Teckin smart plugs. :D

ch3lmi, 2020-06-26:
Hey there,

I certainly second this request and while I fully agree with the  "replace your router" approach, there are multiple cases where this won't work.
In my case I am living in a small apartment in the mountains with nothing else but a 4G network. I have tested multiple routers from various brands but up to now (and admittedly this is a niche market) haven't found one that would keep this on for a longer period of time in a stable way.
The 4G network is shaky, and when the router is connected to the network but has no internet connectivity, well guess what ? only a reboot will fix it.
For me the Wifi "hack" described above won't work, as while there is no internet, the wifi is just running fine.
I would really appreciate a watchdog on a specific IP that would power cycle upon several failures in a row.

For the time being I am using an ubiquity mpower socket that has exactly this feature. Unfortunately this product is long time discontinued and I am looking for a replacement.
The ubiquity implementation will do exactly this with their watchdog, and by following this you  have exactly the needed behaviour:
https://community.ui.com/questions/How-to-create-a-ping-watchdog-to-powercycle-outlets-on-Internet-outage-manual-powercycle-from-cloud/bf354c69-dcc4-4380-8f55-36be79f195fe

I guess implementing something like this should be feasible.
It would probably even be quite easy to port apinger for this purpose: https://github.com/Jajcus/apinger

meingraham, 2020-06-26:
Tasmota has a Ping command now. You can wrap some rules and timers around it and likely create a watchdog.

tyjtyj, 2020-09-11:
> 
> 
> @ch3lmi
> 
> Tasmota has a Ping command now. You can wrap some rules and timers around it and likely create a watchdog.

Do you have some example how this works ? I am very bad with rules... appreciate it.

meingraham, 2020-09-11:
@tyjtyj https://tasmota.github.io/docs/Rules/#watchdog-for-wi-fi-router