arendst/Tasmota #3078: RF bridge: Only first of multiple MQTT commands gets executed

nicolas-fricke, 2018-06-27:
**When receiving multiple MQTT commands at once, only first command seems to be executed.**

I have a setup where I use a _Sonoff RF Bridge 433_ and use the [Home Assistant](https://www.home-assistant.io) [MQTT light component](https://www.home-assistant.io/components/light.mqtt/) along with some [Elro 433MHz power outlets](http://wiki.openpicus.com/index.php/433Mhz_wireless_power_control_with_Wi-Fi) to control multiple lights around the apartment. Switching the individual lights on/off via Home Assistant / MQTT works just fine.

I have grouped three lights into a [light group](https://www.home-assistant.io/components/light.group/) which causes them to be switched together. Using the Tasmota console, I can also see that all three commands have been received by the Tasmota / Sonoff Bridge:
```
00:28:33 MQT: tasmota/sonoff-rf-bridge/stat/RESULT = {"RfCode":"#401454"}
00:28:33 MQT: tasmota/sonoff-rf-bridge/stat/RESULT = {"RfCode":"#401514"}
00:28:33 MQT: tasmota/sonoff-rf-bridge/stat/RESULT = {"RfCode":"#501154"}
```

Yet, only the first socket is switched on/off. I believe this may be due to either of the two issues: a) the three commands are being triggered too closely together and therefore not being interpreted as distinct commands by the remaining two sockets, or b) the bridge simply ignores the last two commands and only sends the first one.

When then manually triggering `RfCode #401514` or `RfCode #501154`, e.g. via the console, the respective lights switch as expected.

- [X] Searched the problem in issues and in the wiki _
  (proved hard since I didn't quite know what to look for)_
- [X] Hardware used: Sonoff Rf Bridge 433 (new version from 2018)
- [X] Provide the output of command ``status 0`` :
```
00:50:55 CMD: status 0
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"sonoff-rf-bridge","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T00:09:26","StartupUTC":"2018-06-26T23:41:29","Sleep":0,"BootCount":19,"SaveCount":54,"SaveAddress":"FA000"}}
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS2 = {"StatusFWR":{"Version":"5.13.1","BuildDateTime":"2018-05-10T01:59:37","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Pica Pica Net",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS4 = {"StatusMEM":{"ProgramSize":515,"Free":488,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff-rf-bridge-4719","IPAddress":"192.168.2.126","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.2.1","Mac":"DC:4F:22:CB:32:6F","Webserver":2,"WifiConfig":3}}
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.2.128","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_CB326F","MqttUser":"mqtt","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS7 = {"StatusTIM":{"UTC":"Tue Jun 26 23:50:55 2018","Local":"Wed Jun 27 00:50:55 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"05:19","Sunset":"20:28"}}
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS10 = {"StatusSNS":{"Time":"2018-06-27T00:50:55"}}
00:50:55 MQT: tasmota/sonoff-rf-bridge/stat/STATUS11 = {"StatusSTS":{"Time":"2018-06-27T00:50:55","Uptime":"0T00:09:26","Vcc":3.394,"Wifi":{"AP":1,"SSId":"Pica Pica Net","RSSI":100,"APMac":"70:3A:CB:9A:14:9C"}}}
```

**To Reproduce**
1. Set up a Tasmota Sonoff Rf Bridge to control multiple 433MHz power outlets
2. Using MQTT, send commands for multiple sockets _at the same time_ (e.g. using Home Assistant and a light group containing multiple MQTT lights)

**Expected behavior**
I would expect that every 433MHz switch reacts to the commands sent to them.

**Actual behavior**
Only the first command received by the bridge seems to have an effect.

**Additional context**
I have already tried using the `delay` command to set a delay between the subsequent commands, but to no avail.

[From the wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#management):

> ### Management
> 
> Command           | Payload    | Description
> ------------------|------------|-------------------------------------------------------------------------------
> Delay             |            | Reset backlog delay to 0.2 seconds
> Delay             | 2..3600    | Set delay between two backlog commands with 0.1 seconds increment

I'd appreciate any pointers towards what the issue might be and how it could be addressed. Thank you :)

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2018-06-27:
You are right, the commands need a delay between sending. RF Power Outlets are to slow....
Command delay is only used for command Backlog!
So i see two possibilitys:
Switch your group with Command Backlog with a Delay setting that works or
do the neccersary delay in HA

Jason2866, 2018-06-27:
You could change your setup for the bridge as follow.
define the switch command ON / OFF always with the command backlog

Sniplet example (HA OpenHAB) needs adaption for HA:

`{ mqtt=">[broker:cmnd/sonoff-bridge/Backlog:command:ON:Delay 2; RfSync 11320; RfLow 360; RfHigh 1090; RfCode #151514; Delay 20; Status7]" }`

Sending RF timing commands and the RFCode fast in order change delay time to 20 and send dummy command after the needed wating time.

nicolas-fricke, 2018-06-27:
Thanks @Jason2866 for your quick reply! I will try to get this to work with Home Assistant :) Though I then probably won't be able to use the [built-in MQTT light component](https://www.home-assistant.io/components/light.mqtt/). 

If anyone has solved this with HA I'd be happy about pointers to how to best add the delay

ascillato2, 2018-07-10:
Hi,

Could you manage to solve your issue?

vkhurana, 2018-07-14:
I had the same issue. worked around it using the suggestion above, but it's far from ideal.

It would be good to have a delay param, or setting.

your homeassistant switch config will look like:
```
- platform: mqtt
  name: some_switch
  command_topic: "cmnd/sonoff/Backlog"
  payload_on: "RfCode 1234; Delay 5;"
  payload_off: "RfCode 1235; Delay 5;"
```

ascillato2, 2018-07-25:
Hi,

@nicolas-fricke 

Have you managed to solve your issue with the workaround explained by @vkhurana ?

ascillato2, 2018-07-28:
Closing as there is no feedback. Solution explained above.

closed 2018-07-28 by ascillato2

sebastianovide, 2019-02-28:
I think that either it is a tasmota bug or a tasmota feature not implemented. The delay on HA is just a workaround. What’s the point in using MQTT if tasmota is not able to consume messages ? Tasmota should 
- 1 - consume a message
- 2 - execute its command (doesn’t matter how long it takes)
- 3 - repeat 

Not sure why this is closed

ascillato, 2019-02-28:
>I think that either it is a tasmota bug or a tasmota feature not implemented.

Please, do some tests and you will find that the RF chip (not the esp8266) requires some time to code/decode. 

>Not sure why this is closed

The issue can't be solved. The restriction is outside Tasmota software.

This github issue was closed because a workaround was given and there were no more feedback from the author of the issue.

Please, if you find a way in the firmware of the rf chip or in Tasmota to behave as you said, do a pull request and we will give you all the support we can to merge your solution. 

Thanks

sebastianovide, 2019-03-02:
Hola @ascillato 

its a pity that you are ignoring a real bug. It is nothing to do with the RF chip, it is just simple logic. Consume only those messages that you are able to manage. 1 MQTT message at the time. That is the whole point of messages systems.

It is your project and if you want to ignore users input it is all up to you. I'm just passing by here and just wanted to provide you some feedback.

Just in case you are interested, check Espurna: I've just tested it and it works exactly as I mentioned.

Adios y buena suerte !

ascillato, 2019-03-02:
Sorry. I think we are not understanding each other.

>its a pity that you are ignoring a real bug. It is nothing to do with the RF chip, it is just simple logic. Consume only those messages that you are able to manage. 

That seems that your issue is different than the one of this thread. Please, contact us in the Tasmota Support Chat for further explanations. I'm not having those problems as you have stated. If I send by RF one command, it is being received in the bridge. If I sent several commands by RF at the same time, not all are received (that depends on the rf chip). In the other way, if I send by mqtt a command to make the rf bridge to transmit a command by rf, it is being executed right away. If I sent several commands without any time spacing between them, some are lost. Some times this is because the rf chip is sending data and ignoring any other command and some other times is because Tasmota is already processing and it is ignoring other commands.

When we said that if you find a better way in the code to manage things, we meant it that. We love receiving pull requests with any enhancement (that is what open source is about). This project is made by all the work shared in the github community. It is free and supported by several people in their free time. 

>It is your project and if you want to ignore users input it is all up to you.

It isn't my project. It is Theo's. I'm just a helper here. It would be easier for me to just not answering to you when you have first posted and I would be saving myself of some bitching. Be aware that I'm not ignoring you. I wanted only to explain to you what you have asked. 

We would need to see the output of your Tasmota console (using weblog 4) when you see that behaviour. May be you are having other issue. For example, Tasmota is being offered compiled with 3 different cores. Some cores work better in some routers than others. So, may be you are using Tasmota with a core that is not 100% compatible with your router, so, sometimes the core hungs for a couple of seconds if doing DNS searching or multicast registering. That can be seen in the console. The solution for that is testing other cores. These other problems are outside Tasmota and are SDK issues (SDK is closed source) 

So, may be, you have used Tasmota with a core that doesn't work fine with your router but you have used espurna with a core that works fine.

Hope you can share more information in the Tasmota Support Chat in order to address your issue.

Jason2866, 2019-03-02:
Please try a actual version. Your used version 5.13.x is very old and out of any support...