arendst/Tasmota #3080: Androids app MQTT DASHBOARD config not found.

centiane, 2018-06-27:
Hi, i need help for start APP binding with sonoff Tasmota Dual. 
This is mi config MQTT in sonoff Tasmota.
[image]

is ok?

and this is mi Broker
[image]

This send a mqtt message but not receiver the sonoff via websockets.
[image]

I am four day loss and not find the problen, help me please.

how activate a power1 on?
Is posible activate a relay whitout local mosquito broker, only a public broker same https://api.cloudmqtt.com? Explained me who is this? Thank you.
I am a novice, and sorry for my english.
Regards Centiane.

Frogmore42, 2018-06-27:
Log from Tasmota especially with output from Status 0

Many people have trouble with public brokers, not sure why this might be.  I have never tried to use one, since it is really easy to set up a local one, even just to test things.  It literally takes less than 30min to download mosquitto to a laptop and start it and play with publishing and subscribing.  You can then configure Tasmota to use it.  Once you understand how things work better, it would probably be easier to understand why it is not working for the cloud ones.

Frogmore42, 2018-06-27:
But, it looks like the user name in your broker and in Tamotsu don't match.  The log will tell you more about what is happening.

centiane, 2018-06-29:
Thank for reply.
This is a User config
[image]

This a info sonoff
[image]
And this a console sonoff
[image]
The config think is ok, but not found mqttcloud, why?
I am not undertand this error.
Help me please. regards centiane.

Frogmore42, 2018-06-30:
The part of the log indicates it was connected to MQTT.  It also indicates there was a fatal exception before this.  That can be because of bad hardware or power.

Still missing the output from status 0, which might provide more information.

centiane, 2018-07-01:
Hi Frogmore42
"Still missing the output from status 0, which might provide more information." this is the status .
The excepcion is for disconect and connect the power supply for restart.
The conection is ok, but not respond a mqtt from broker in websoket console or i am not send the correcty command for change the output status.
ON ----> POWER1 ON
OFF -----> POWER1 OFF.
This comand fron sonoff console found pk, but the broker websockct console not found.
Can you say me the correct format for send the command?
Thank you.
Regards Centiane.

Frogmore42, 2018-07-01:
Did you see the Wiki on this:
https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview

What exactly are you typing, and in what?

It sounds like you are able to get the command to work from the console.
But, when you try and send the command from MQTT it doesn't work?

That is probably because you are not sending the command to the correct destination.
When you use MQTT you have to specify the topic correctly, for your example that would probably be:
cmnd/cumbresmayores/power
The payload then needs to be: on

centiane, 2018-07-02:
Thank Frogmore42.
I read a MQTT-overview and i testing for all format.
[image]
But not response a MQTT command, i dont think any idea.
Only this error is me. I testing for all format and other broker, i open a port in routers, only not testing a broker in my local lan.
Whats is the problen?
Regards Centiane.

closed 2018-09-01 by stale[bot]