arendst/Tasmota #3083: Commands ADD, SUB, MULT, etc only use integers

ascillato, 2018-06-27:
**Describe the bug**
_A clear and concise description of what the bug is._

The following commands only support integers. When using values with decimals, those are truncated to integers:

* `ADDx`
* `SUBx`
* `MULTx`
* `SCALEx`
* `KNXTX_VALx`

I'm informing this to be taken into account. I will work on this to try to support decimals. Any Help or suggestion is welcome.

_Also, make sure these boxes are checked [x] before submitting your issue - Thank you!_

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : **NodeMCU and some Sonoffs**
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE

17:48:41 MQT: stat/cocina/STATUS = {"Status":{"Module":18,"FriendlyName":["Cocina"],"Topic":"cocina","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
17:48:41 MQT: stat/cocina/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T04:41:27","StartupUTC":"2018-06-27T16:07:14","Sleep":0,"BootCount":46,"SaveCount":169,"SaveAddress":"3FB000"}}
17:48:41 MQT: stat/cocina/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-12T22:44:43","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:48:41 MQT: stat/cocina/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NetWireless","NetWireless"],"TelePeriod":300,"SetOption":["0E008009","55818000"]}}
17:48:41 MQT: stat/cocina/STATUS4 = {"StatusMEM":{"ProgramSize":523,"Free":2548,"Heap":11,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00002C0A","1F90A984","0C000010","01021792","00000000"]}}
17:48:41 MQT: stat/cocina/STATUS5 = {"StatusNET":{"Hostname":"cocina-1757","IPAddress":"192.168.1.6","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:F7:A6:DD","Webserver":2,"WifiConfig":5}}
17:48:41 MQT: stat/cocina/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F7A6DD","MqttUser":"homeassistant","MqttType":2,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
17:48:41 MQT: stat/cocina/STATUS7 = {"StatusTIM":{"UTC":"Wed Jun 27 20:48:41 2018","Local":"Wed Jun 27 17:48:41 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"08:51","Sunset":"18:18"}}
17:48:41 MQT: stat/cocina/STATUS10 = {"StatusSNS":{"Time":"2018-06-27T17:48:41"}}
17:48:41 MQT: stat/cocina/STATUS11 = {"StatusSTS":{"Time":"2018-06-27T17:48:41","Uptime":"0T04:41:27","Vcc":2.784,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"NetWireless","RSSI":52,"APMac":"18:D6:C7:80:38:2C"}}}

```

**To Reproduce**
_Steps to reproduce the behavior:_

```
17:50:52 CMD: var1 12.4
17:50:52 MQT: stat/cocina/RESULT = {"Var1":"12.4"}
17:50:58 CMD: add1 3.2
17:50:58 MQT: stat/cocina/RESULT = {"Add1":"15"}
17:51:08 CMD: var1
17:51:08 MQT: stat/cocina/RESULT = {"Var1":"15"}
```

**Expected behavior**
_A clear and concise description of what you expected to happen._

That the values with decimals were not truncated to integers

**Screenshots**
_If applicable, add screenshots to help explain your problem._

None

**Additional context**
_Add any other context about the problem here._

None

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-06-27:
There is a dedicated (small code footprint) function for this. Don't remember the name but it's in support.ino.

Once found look for code examples already in tasmota

ascillato, 2018-06-27:
Great. Will look for it. Thanks!

ascillato, 2018-06-28:
So as to use decimal values in the `SCALE` command, I had to made a new approach to search for values in between ",". I made the `subStr` function to search for values. I could not use the same approach you use at `TIMEDST` command.

If you think that this new search function could be useful for another routines, we can move it to _support.ino_

Also, if you think that can be improved in any way, I'm willing to help.

closed 2018-06-28 by ascillato

misc2000, 2018-07-01:
I did make some tests with multi DS18B20 sensors and can confirm that decimal values for KNXTX_VALx work.

ascillato, 2018-07-01:
Excelent! Thanks :+1:

misc2000, 2018-07-01:
but could it be that you only send 1 digit after the dot with KNXTX_VALx?
I would prefer to have 2 digits like with the direct sensor value send which is always 2 digits after the dot on the knx side...

ascillato, 2018-07-02:
Hi @misc2000 

KNXTX_VALx is sending 2 decimal digits. Can you try again?

My console output (sending):

[image]

My Home Assistant (receiving):

[image]

misc2000, 2018-07-02:
I did run one unit with 2 x DS18B20 temperature sensors for 24h. 
Sensor1 is sending with the build in function to group 6/2/1 and sensor2 is sending with this rule to 6/2/2:
`rule on tele-DS18B20-2#temperature do backlog knxtx_val1 %value%; var1 %value% endon on event#knxrx_req1 do knxtx_val1 %var1% endon on Clock#Timer=16 do Restart 1 endon`
Both are shown only with 1 digit in the console log, but sensor1 is sending with 2 digits to the KNX bus, see 2nd screenshot and sensor2 not.
My console output 
[image]
here a log off a knx group monitor of the same time
[image]

As you can see sensor1 is with 2 digits and sensor2 only rounded after the 1 digit.

Here also a record of the values of this 2 KNX groups from the last hours:
```
date       time 	     sensor1  sensor2
2018-07-02 10:00:00	30.7600	30.8000
2018-07-02 10:01:00	30.7600	30.8000
2018-07-02 10:02:00	30.7600	30.8000
2018-07-02 10:03:00	30.7600	30.8000
2018-07-02 10:04:00	30.7600	30.8000
2018-07-02 10:05:00	30.7600	30.8000
2018-07-02 10:06:00	30.7600	30.8000
2018-07-02 10:07:00	30.7600	30.8000
2018-07-02 10:08:00	30.7600	30.8000
2018-07-02 10:09:00	30.7600	30.8000
2018-07-02 10:10:00	30.7600	30.8000
2018-07-02 10:11:00	30.7600	30.8000
2018-07-02 10:12:00	30.7600	30.8000
2018-07-02 10:13:00	30.7600	30.8000
2018-07-02 10:14:00	30.7600	30.8000
2018-07-02 10:15:00	30.7600	30.8000
2018-07-02 10:16:00	30.7600	30.8000
2018-07-02 10:17:00	30.7600	30.8000
2018-07-02 10:18:00	30.7600	30.8000
2018-07-02 10:19:00	30.7600	30.8000
2018-07-02 10:20:00	30.7600	30.8000
2018-07-02 10:21:00	30.7600	30.8000
2018-07-02 10:22:00	30.7600	30.8000
2018-07-02 10:23:00	30.7600	30.8000
2018-07-02 10:24:00	30.7600	30.8000
2018-07-02 10:25:00	30.7600	30.8000
2018-07-02 10:26:00	30.7600	30.8000
2018-07-02 10:27:00	30.7600	30.8000
2018-07-02 10:28:00	30.7600	30.8000
2018-07-02 10:29:00	30.7600	30.8000
2018-07-02 10:30:00	30.7600	30.8000
2018-07-02 10:31:00	30.7600	30.8000
2018-07-02 10:32:00	30.7600	30.8000
2018-07-02 10:33:00	30.7600	30.8000
2018-07-02 10:34:00	30.7600	30.8000
2018-07-02 10:35:00	30.7600	30.8000
2018-07-02 10:36:00	30.7600	30.8000
2018-07-02 10:37:00	30.7600	30.8000
2018-07-02 10:38:00	30.7600	30.8000
2018-07-02 10:39:00	30.7600	30.8000
2018-07-02 10:40:00	30.7600	30.8000
2018-07-02 10:41:00	30.7600	30.8000
2018-07-02 10:42:00	30.7600	30.8000
2018-07-02 10:43:00	30.7600	30.8000
2018-07-02 10:44:00	30.7600	30.8000
2018-07-02 10:45:00	30.7600	30.8000
2018-07-02 10:46:00	30.7600	30.8000
2018-07-02 10:47:00	30.7600	30.8000
2018-07-02 10:48:00	30.7600	30.8000
2018-07-02 10:49:00	30.7600	30.8000
2018-07-02 10:50:00	30.7600	30.8000
2018-07-02 10:51:00	30.7600	30.8000
2018-07-02 10:52:00	30.7600	30.8000
2018-07-02 10:53:00	30.7600	30.8000
2018-07-02 10:54:00	30.7600	30.8000
2018-07-02 10:55:00	30.7600	30.8000
[... 615 lines cut ...]
2018-07-02 21:11:00	32.8800	32.7000
2018-07-02 21:12:00	32.8800	32.7000
2018-07-02 21:13:00	32.8800	32.7000
2018-07-02 21:14:00	32.8800	32.7000
2018-07-02 21:15:00	32.8800	32.7000
2018-07-02 21:16:00	32.8800	32.7000
2018-07-02 21:17:00	32.8200	32.6000
2018-07-02 21:18:00	32.8200	32.6000
2018-07-02 21:19:00	32.8200	32.6000
2018-07-02 21:20:00	32.8200	32.6000
2018-07-02 21:21:00	32.8200	32.6000
2018-07-02 21:22:00	32.8200	32.6000
2018-07-02 21:23:00	32.8200	32.6000
2018-07-02 21:24:00	32.8200	32.6000
2018-07-02 21:25:00	32.8200	32.6000
2018-07-02 21:26:00	32.8200	32.6000
2018-07-02 21:27:00	32.7600	32.6000
2018-07-02 21:28:00	32.7600	32.6000
2018-07-02 21:29:00	32.7600	32.6000
2018-07-02 21:30:00	32.7600	32.6000
2018-07-02 21:31:00	32.7600	32.5000
2018-07-02 21:32:00	32.7600	32.5000

```
Also the same 2nd Value only with 1 digit :-(

ascillato, 2018-07-03:
Will investigate

ascillato, 2018-07-03:
Hi, when looking at the teleperiod message, in the json message can be seen that the value is already there with just 1 decimal. This is being done by the DS18B20 driver. So for making it display 2 decimal digital, just go to the console and type the command `TempRes 2`

misc2000, 2018-07-03:
works !
Did not know that command before.
But strange that it was working different for the 2 sensors before.
Anyway thanks for the hint