arendst/Tasmota #3084: Settings in user_config.h do not take effect

AnderssonPeter, 2018-06-27:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [Currently down?] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : Wemos d1 mini clone

```
Executing task: C:\Users\Peter.Asgard\.platformio\penv\Scripts\platformio.exe device monitor <

--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
{d␀d��|␀�l�|␂␌␌␄�␄d�␌c|��␃�␓�{�c�␌c��'n�doo���␌c␜8��dslslx�g�␘␃␄␄�␄l␌��␌␌␄b␄o�|␃l�␌␌�b��oo�␀l��l ␃�␒␓gg␄d`␃␏␃o;���g␄␄c␄�␎l␇r��g␄␄c␄�␎l�r␓���␛d�l`␂��n�␃�
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_066171, GroupTopic sonoffs) Version 5.14.0-2_3_0
00:00:00 WIF: Connecting to AP1  in mode 11N as sonoff-0369...
00:00:11 WIF: Connect failed with AP timeout
00:00:11 WIF: Connecting to AP1  in mode 11N as sonoff-0369...
00:00:22 WIF: Connect failed with AP timeout
00:00:22 WIF: WPSConfig active for 3 minutes
00:00:42 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
vc9f0c112
~ld
�
```

And i have the following settings in user_config.h
```ini
// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "192.168.2.254"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.2.27"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "P&P 2.4GHz"                // [Ssid1] Wifi SSID
#define STA_PASS1              "qwerty"                // [Password1] Wifi password
#define STA_SSID2              "P&P 5GHz"                // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "qwerty"                // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_MANAGER     // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)
```
**(Please, remember to close the issue when the problem has been addressed)**
Correct me if I'm wrong but shouldn't my wifi name be listed in the `00:00:00 WIF: Connecting to AP1  in mode 11N as sonoff-0369...` message? and when it fails it prints `00:00:22 WIF: WPSConfig active for 3 minutes` but i picked `WIFI_MANAGER`.

ascillato, 2018-06-27:
Hi

Please read the CFG_HOLDER behaviour at the WIKI:

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#config-problems-can-cause-boot-loops-items-set-in-user_config-to-not-appear-etc

https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#cfg_holder-explained

Also, remember that the ESP8266 chip (sonoff devices etc, wemos, etc) don't support WIFI at 5Ghz

If this solves your issue, please close it. Thanks :+1:

closed 2018-06-28 by ascillato2

AnderssonPeter, 2018-06-28:
Thanks it solved the problem!