arendst/Tasmota #3085: Wemos d1 mini battery shield

Gary202, 2018-06-27:
**Have you look for this feature in other issues and in the wiki?**

**Is your feature request related to a problem? Please describe.**
_Not A problem._

**Describe the solution you'd like**
_Support the wemos d1 battery shield with battery voltage monitor with analog pin A0._

**Describe alternatives you've considered**
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**
_It would be awesome to have the analog pin A0 to monitor the voltage of the wemos d1 battery shield with voltage  in the mqtt message ._

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-06-28:
Hi

The Analog pin is already supported and working.

You just need to comment [the line 253 in user_config.h](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/user_config.h#L253) file, in order to let Tasmota use the A0 pin as Analog source.

So, for monitoring the voltage you will have the reading as a sensor by MQTT (without the need to configure anything else than WIFI and MQTT server credentials)

If this solves your issue, please close it. Thanks :+1:

Gary202, 2018-06-28:
What is the input voltage range?

ascillato, 2018-06-28:
0v - 3.3v

Gary202, 2018-06-28:
Is 3.3 the max. So if I was going to use it to monitor the battery voltage of a lipo I should use a voltage divider?

ascillato, 2018-06-28:
As it is a shield, should be ready to be used in the wemos.

According to the wemos wiki site (https://wiki.wemos.cc/products:d1_mini_shields:battery_shield), in the schematics has a connection by a jumper to the battery already prepared for the wemos.

So, you should check, but seems that it will work.

ascillato, 2018-06-28:
Hi,

If your question is addressed, please close this issue. Thanks! :+1:

closed 2018-06-28 by Gary202