arendst/Tasmota #3086: More switches

Lawrencezarb, 2018-06-28:
I am working on a project using a D1 mini and Tasmota.

I see that I can have upto 8 Relays configured, but it only allows 4 switches to be configured.

How can I have more switches available, I actually need 5.!

arendst, 2018-06-28:
Working on it (#2885) to increase to 8

Lawrencezarb, 2018-06-28:
Thanks, I tried to increase the MAX_SWITCHES but this had no effect.

eXCepT74, 2018-09-08:
Hello
 I working on my smarthouse installation and decided to use tasmota firmware because succesfully built many smart devices for use on my own. Similar to Lawrencezarb i use D1 mini and need more gpio's to use - and here my question. In my projest i use many sensors ( windows, doors contact sensors , motion etc ) and need up to 8 switches to use. But i noticed that if use for example switches 1-8 everyone publishing POWER if i add 3 relays some will publishing POWER 1 some POWER 2 or POWER 3. That what i need to up to 8 independent inputs that will publish POWER 1-8. Is that posible to do in some config way ? The best option is to make posibility independent control of gpios . In other words if i set gpio 1 ,2 and 3 to switch without any relays - i should can do read POWER1-3 depend gpio in use.
Regards ! Good job anyway !