arendst/Tasmota #3090: How to invert Sonoff Touch logic Normally Closed

leonardcr, 2018-06-28:
Hi, I'm using a sonoff touch to control a magnetic door. I was able to flash tasmota into Sonoff Touch, and I configured the Power to show CLOSED and OPEN instead of ON/OFF.
My problem is that when I use PulseTime to add 5s timer, it only works in a way 
OPEN->CLOSED->5s->OPEN. 
I was not able to change the way CLOSED->OPEN->5s->CLOSED.
Changing PulseTime while the Touch is OPEN or CLOSED has no difference in the behaviour. 
I tried change SwitchMode, but any option from 1 to 4 did what I needed.
Is that a way to do that ?

Thanks.

Frogmore42, 2018-06-28:
It's not clear what you want.  Pulsetime is normally used to limit the amount of time the relay is energized.  For a Sonoff device this means the relay contacts are connected/closed.  However, it is possible to invert the signal to the relay so that the opposite happens.  This would mean that normally the relay would be energized and only for the duration of pulsetime would it be deenergized.  Is that what you want?

leonardcr, 2018-06-28:
In fact, is the contrary. Since the magnectic lock is locked when energized, I would like to let the sonoff touch energized normally (so normally locked and energized) and when I press the button, it would deenergized and after pulsetime, it would energized again.

Frogmore42, 2018-06-29:
You need to invert the relay signal.  This means you need to select the module type as generic and then select the GPIO pins to match what the Touch uses, just invert the one for the relay.  This will actually turn the relay on when the Sonoff thinks it is off.  So, pulsetime will work the way you want.

leonardcr, 2018-06-29:
Hi, worked like a charm. Just to let saved and detailed here for others that could need to do the same.
We have to go into configuration -> configure module, chose 18 Generic.
For sonoff touch we have to change:
D3 GPIO0 -> 13 Button1
D6 GPIO12 -> 25 Relay1i
D7 GPIO13 -> 48 Led1
Now save, restart and that's it, the logic will be inverted.
Thanks a lot @Frogmore42 for your help on this.

closed 2018-06-29 by leonardcr

mbt45, 2018-11-02:
hi leonardcr

How did you change configuration of your sonoff to get CLOSED and OPEN instead of ON/OFF?

mbt45, 2018-11-03:
Thank you very much. I'll try to get it done there.