arendst/Tasmota #3091: How can i enable connectivity to app sonoff pow r2

albionbeka, 2018-06-28:
I dont know how to enable connectivity to my own app and how can the app scan for sonoff pow r2

ascillato, 2018-06-28:
Hi,

In the wiki is all the information about REST API, MQTT, and KNX.

You also can use Yeti app for your phone.

Remember that Tasmota is a firmware for sonoff devices that allow connection to MQTT for home automation.

ascillato, 2018-06-29:
Your question is addressed? You need something else?

If not, please close this issue. Thanks :+1:

albionbeka, 2018-06-29:
I need to make my own tho i need to make it in albanian language thats why

ascillato, 2018-06-29:
Ok, I understand. You want to make your own. Perfect. So, what else do you want?

If the only thing is the translation to albanian, talk to @luisfpinto and his team that are developing the phone app called Yeti, that it is very good and well designed app. If you want, you can help them with the translation to albanian.

Also here, Tasmota has several translations but Albanian is missing and also you can help on that if you want.

albionbeka, 2018-06-29:
And to make some other features

ascillato, 2018-06-29:
Ok, Cool.

Please, keep us posted. Thanks :+1:

closed 2018-06-29 by ascillato2