arendst/Tasmota #3093: 4 switches

servim, 2018-06-29:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : d1 mini
- [x] Provide the output of command ``status 0`` :
```
13:10:40 CMD: status 0
13:10:40 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:10:40 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:36:35","StartupUTC":"2018-06-29T04:34:05","Sleep":0,"BootCount":17,"SaveCount":55,"SaveAddress":"3F5000"}}
13:10:40 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-27T16:00:02","Boot":6,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
13:10:40 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["svm",""],"TelePeriod":60,"SetOption":["00008009","55818000"]}}
13:10:40 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":511,"Free":2560,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3,"Features":["00000419","0F80A790","00000000","0600009A","00000000"]}}
13:10:40 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2951","IPAddress":"192.168.1.78","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:5A:0B:87","Webserver":2,"WifiConfig":3}}
13:10:40 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.14","MqttPort":1974,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5A0B87","MqttUser":"svm-MQTT-logIn","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
13:10:40 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 29 05:10:40 2018","Local":"Fri Jun 29 13:10:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":8,"Sunrise":"04:43","Sunset":"21:30"}}
13:10:40 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-06-29T13:10:40","Switch1":"OFF","Switch2":"OFF","Switch3":"OFF","Switch4":"OFF","DS18B20":{"Temperature":24.0},"TempUnit":"C"}}
13:10:40 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-06-29T13:10:40","Uptime":"0T00:36:35","Vcc":2.956,"Wifi":{"AP":1,"SSId":"svm","RSSI":82,"APMac":"32:B2:DC:5C:00:A0"}}}
13:11:20 MQT: tele/sonoff/STATE = {"Time":"2018-06-29T13:11:20","Uptime":"0T00:37:15","Vcc":2.956,"Wifi":{"AP":1,"SSId":"svm","RSSI":82,"APMac":"32:B2:DC:5C:00:A0"}}
```
Hello!
I want to use this firmware with the d1 mini module, to track four switches (D5, D6, D7, D2). My settings:
{"SwitchMode1": 1}
{"SwitchMode2": 1}
{"SwitchMode3": 1}
{"SwitchMode4": 1}
{"SwitchTopic": "switch"}
When switching any of the four buttons, I see the same result:
13:06:42 MQT: cmnd / switch / POWER = ON
13:06:45 MQT: cmnd / switch / POWER = OFF
How can I tell which of the four switches sent a message?
Thank you!

arendst, 2018-06-29:
Do not use switchtopic but use rules instead.

As an example ``rule on switch1#state do publish cmnd/theo1/power %value% endon on switch2#state do publish cmnd/theo2/power %value% endon``

servim, 2018-06-29:
Thanks for the quick response!
Only I do not understand where to write the rules.
in a user_config.h
#define USE_RULES

arendst, 2018-06-29:
See wiki > Using Rules

ascillato, 2018-06-29:
Rules are typed on the console of the Tasmota webmenu

ascillato2, 2018-06-29:
Closing. Feel free to reopen if needed. Thanks :+1:

closed 2018-06-29 by ascillato2