arendst/Tasmota #3094: GND and switch1  will make d1 stuck in reboot loop

sam9s, 2018-06-29:
**Describe the bug**

_A clear and concise description of what the bug is._

Hi developer

there is a very strange issue I am facing and i am not getting any ans. maybe you can help.  this is how I have connected  D2 is relay one and  D4 switch1, now when I short D4 to GND, the switch works as intended, but if I keep the switch on (i.e D4 and ground is shorted) and in that situation I reboot the device , it gets stuck in loop and does not get rebooted, and I loose the web interface. How ever if I disconnect the D4 and ground and then reboot all is ok and device reboots and I have the web interface. . Basically if GHD and Switch1 are ON (shorted) and in that situation/condition device reboots (or power goes down and comes back) . D1 Mini never boots and no web interface. .. I have to unshort (disconnect) GND and switch1 and then reboot the device again to get the web interface back ........ .Why is this. ???? is there a solution.....can you address this problem .... please do try to re create and tell me if you also face the same issue and how you managed to rectify it.

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues and in the wiki
- [ ] Hardware used : D1 Mini and tasmota
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**To Reproduce**
_Steps to reproduce the behavior:_

**Expected behavior**
_A clear and concise description of what you expected to happen._

**Screenshots**
_If applicable, add screenshots to help explain your problem._

**Additional context**
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Leseratte10, 2018-06-29:
This is not caused by Tasmota. The ESP8266 needs GPIO0 and GPIO2 to be high (= not connected to anything, or connected to 3.3V) during boot. GPIO0 and GPIO2 are the pins D3 and D4 on the Wemos D1 mini. D3 and D4 should only be used to connect push buttons (which are only low when pressed), not toggle switches. 

Just use another pin instead of D3/D4.

Jason2866, 2018-06-29:
Plz close issue. THX

ascillato, 2018-06-29:
Added this clarification to the wiki at

https://github.com/arendst/Sonoff-Tasmota/wiki/Control-a-Sonoff-using-a-remote-button

Thanks.

closed 2018-06-29 by ascillato2

sam9s, 2018-06-30:
OK ,,.. thanks for the reply  ... You solution worked ... what I did was not use D3 or D4 for either switch or relay and I was able to boot the devvice even if the switch and GND were shortened. I  stated thinking if this is a GPIO Pins issue let me see what works and what not, and then I started experimenting with all Pins both as switch and relay. Below are my findings ..... 

On D1 Mini we have following switches ..
D0, D1, D2 .... till D8
Rx and Tx

D3 and D4 does not work  (as pointed by you)
D0 and D8 also does not work as switch but they work as relay BUT when switch1 (any GPIO) is shortened with GND, device does not boots and no web interface is there.
Tx and Rx works as switch and relay both BUT again when switch1 (any GPIO) is shortened with GND, device does not boots and no web interface is there.
That leaves us with D1, D2, D5, D6 and D7
all the above works perfectly, use any of the above GPIOs as switch or relay, the device works and boots nicely as well even if any one of the GPIO as switch is shortened with GND and rebooted.

So my conclusion is with D1 mini use only D1,2,5,6, or 7 for your switches and relays ....

Let me know if my findings makes any scene and what might be the technical reason behind it....

And thanks for all the help ..

Regards
Sammy