arendst/Tasmota #3096: Concatenate triggers with logical operators

renne, 2018-06-29:
**Have you look for this feature in other issues and in the wiki?**
Yes.

I suggest to add support for concatening rule triggers with logical operators, e.g.
`on APDS9960#Ambient<100 && switch1#state=1 do power on endon`
to allow a motion detector to switch the light on only when ambient light is below 100 lux.

ascillato, 2018-06-29:
Hi,

I like your suggestion :+1: but remember that every `on [something]` are triggers, not comparators.

For comparators should be needed an IF statement that currently is not implemented in Tasmota.

At this moment you can make that concatenation like this:

```
on switch1#state=1 do var1 100 endon
on switch1#state=0 do backlog var1 0; power1 off endon
on APDS9960#Ambient<%var1% do power1 on endon
```

This means that when the switch is on, the light will turn on but only when you have less than 100 lux in that room. While if the switch is off the light will be off. _(I like your rule idea)_

----------------------------------------------

_For anyone else reading this: the rule need to be all together to work like this:_

```
rule 1
rule on switch1#state=1 do var1 100 endon on switch1#state=0 do backlog var1 0; power1 off endon on APDS9960#Ambient<%var1% do power1 on endon
```

------------------------------------------------

Hope this helps.

ascillato, 2018-06-29:
I will add this example to the wiki

ascillato2, 2018-07-10:
Hi,

If you have further comments or questions about this, please, reopen this issue. Thanks :+1:

closed 2018-07-10 by ascillato2