arendst/Tasmota #3097: Sonoff Bridge with door/windows sensor only sends MQTT message when opened

rhannink, 2018-06-29:
Hello,

I flashed Tasmota firmware 5.14.0 to my RF bridge and paired two different Sonoff door/windows RF sensors with it (learned in at key1  en Key2).

When I open the sensor (remove them apart) I get a MQTT message from Tasmota, but not when I close the sensor.

According to: [another issue](https://github.com/arendst/Sonoff-Tasmota/issues/1736) the sensor sent a message on open, close, tamper and low battery

Gr. Remco

rhannink, 2018-06-29:
Sorry

Wrong assumption. The Sonoff DW1 I was talking about, only sent a signal on opening, leaving the pretty much useless. The other item mentions another door/window alarm sensor.

Issue can be closed

Jason2866, 2018-06-29:
You are the owner of the issue, you have to close.
THX

closed 2018-06-29 by rhannink