arendst/Tasmota #3098: If more than 4 relay -> there is no HueEmulation for Relay5

Jason2866, 2018-06-29:


ascillato, 2018-06-29:
Hi,

I also want to add to your report that from 5 to 8 relays, there is no friendly name option.

reloxx13, 2018-06-29:
this is easy, settings.h

#define MAX_FRIENDLYNAMES      4            // Max number of Friendly names

the max hue lamps is = max friendlynames if more than 4 relaises are connected.

Jason2866, 2018-06-29:
@reloxx13  Thank you. Will try tomorrow.

Jason2866, 2018-06-30:
Change #define MAX_FRIENDLYNAMES 4 // Max number of Friendly names to 8 (in sonoff.h)
doesnt work. One side effect on upgrade Config of device is lost/corrupted.

[image]

There is no Friendly Name 5 and it cant be stored one. T
here is no device "5" when searching with Alexa for the emulated HUE Device.

[image]

arendst, 2018-06-30:
First, by changing define MAX_FRIENDLYNAMES you'll screw up the settings area as noted in the same file (sonoff.h) line 39: ``// Changes to the following MAX_ defines will impact settings layout``

Second, try setting friendlyname5 using the command friendlyname5 as currently a friendlyname above 4 will not be set by the webserver (See xdrv_02_webserver.ino lines 1079 .. 1089)
EDIT: Won't work either as I see I have it fixed to 4 (ha, ha)

Thx for testing. Will make some changes and let you know.

arendst, 2018-06-30:
Try again.

Jason2866, 2018-06-30:
Hallo Theo,

thank you :-) 

Looks good. Friendly name appears and can be changed and stored.
Is there a chance to get a version where the settings arent lost if friendly name is changed from 4 to 8?
Or could you change it general to 8 and keep it compatible?

Alexa find now Relay 5 and it can be switched via voice on and off "Alexa Test an" and "Alexa Test aus"

[image]

arendst, 2018-06-30:
Glad it works.

The problem with supporting this is that I have to reconfigure the settings area. This has been done before by copying the current settings to another area and adding the new ones. Some users tend to move up and down versions and this automagically settings change feature will corrupt their settings. They complain about upgrade problems and I find it hard to explain.

Another problem is that four new friendlynames need a total of 4 x 33 bytes in the settings area which becomes tight.

For now I keep it to four. Might change in the future.

Boris16, 2018-07-31:
Hello,
I want to use the 4 GPIOs displayed in the configuration (GPIO 0,1,3,9) on the Sonoff Dual R2, not the onboard relays. Unfortunately, "friendly name 1" and "friendly name 2" are connected to the two onboard relays. So I can only use 2 more GPIOs as "friendly name 3" and "friendly name 4". Can I change that so I can use all 4 (external) GPIOs 0,1,3 and 9?

Jason2866, 2018-07-31:
Dont use the modul Sonoff Dual R2. Use modul generic instead. Just define every GPIO like you want.

Boris16, 2018-07-31:
Thank you for your prompt reply. I tried this. Unfortunately, GPIO 9 is not displayed.

Jason2866, 2018-07-31:
Yes, correct because on nearly all devices GPIO 9 cant be used.
GPIO 9 is technically only useable on ESP8285 devices. Theo did a (clever) decision to hide this
not to get unnecessary support problems
You can edit the module "generic" in sonoff_template.h to
{ "Generic",         // Any ESP8266/ESP8285 device like WeMos and NodeMCU hardware (ESP8266)
     GPIO_USER,        // GPIO00 D3 Wemos Button Shield
     GPIO_USER,        // GPIO01 TX Serial RXD
     GPIO_USER,        // GPIO02 D4 Wemos DHT Shield
     GPIO_USER,        // GPIO03 RX Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 D2 Wemos I2C SDA
     GPIO_USER,        // GPIO05 D1 Wemos I2C SCL / Wemos Relay Shield (0 = Off, 1 = On) / Wemos WS2812B RGB led Shield
     0, 0, 0,  // Flash connection
     GPIO_USER,      // GPIO09   ONLY on ESP 8285 possible!!
      0, 0, // Flash connection
     GPIO_USER,        // GPIO12 D6
     GPIO_USER,        // GPIO13 D7
     GPIO_USER,        // GPIO14 D5
     GPIO_USER,        // GPIO15 D8
     GPIO_USER,        // GPIO16 D0 Wemos Wake
     GPIO_ADC0         // ADC0   A0 Analog input
  },

Boris16, 2018-08-01:
It works perfectly.
I also made the following changes:
 `  { "Sonoff Dual R2",  // Sonoff Dual R2 (ESP8285)
     GPIO_USER,        // GPIO00 Button 0 on header (0 = On, 1 = Off)
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     GPIO_USER,  // GPIO_REL2,        // GPIO05 Relay 2 (0 = Off, 1 = On) <========
     0, 0, 0,          // Flash connection
     GPIO_USER,        // GPIO09 Button 1 on header (0 = On, 1 = Off)
     GPIO_KEY1,        // GPIO10 Button on casing
     0,                // Flash connection
     GPIO_USER,  // GPIO_REL1,        // GPIO12 Relay 1 (0 = Off, 1 = On) <========
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },
`
So I can configure the Sonoff Dual R2 as a "Sonoff Dual R2"  ;-)
@Jason2866: Once again, thank you very much for your help.

Jason2866, 2018-08-01:
Your welcome! Please close this issue. Thank you.

Boris16, 2018-08-01:
Closes #3098

Jason2866, 2018-08-01:
Sorry, it is already closed!