arendst/Tasmota #3099: Sonoff Basic GPIOs with simple contactor

MikhaelGerbet, 2018-06-29:
Hi !

How i can use a contactor (like a opening detector) with a Sonoff basic R2 and know the contactor state?

I have tested with a GPIO14 like Inter2 and the GPIO1 like a Relay2. It's work but it's not stable because if you restart the sonoff with the contactor state = OFF, the relay is egal to OFF BUT  if you restart the sonoff with the contactor = ON, the relay is egal to OFF and the states are reverted ...

My questions are : 
it is possible to use juste one GPIO like a "contactor" without an other GPIO like a Relay and know by mqtt if his state is egal to ON or OFF (real value) ? 

if no, it is possible to add this feature ? maybe with a voltage listener on the GPIO ? if  volt  on GPIOx === 0 ? 'OFF' : 'ON'.

This feature is verry important for the opening detector (for exemple).

Thank for your help.

ascillato, 2018-06-29:
Please read the wiki at:

https://github.com/arendst/Sonoff-Tasmota/wiki/Control-a-Sonoff-using-a-remote-button

https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs

Thanks

MikhaelGerbet, 2018-06-29:
@ascillato you make my day ! thx !

closed 2018-06-29 by MikhaelGerbet