arendst/Tasmota #3101: [Feedback wanted] Alexa issues after 6.0.0b WIFI Test changes

reloxx13, 2018-06-29:
Anyone else getting Alexa Problems ("Device does not respond") since 6.0.0b?

After this commit:
https://github.com/arendst/Sonoff-Tasmota/commit/20a53950ff208314cb609cb3ecf640c4bf5c4d85

Please provide Feedback 👍 

---

Before this commit Alexa had no issues with the devices, now she tells me often the device does not respond.
After 2-3 times giving the command to power on the device, it will power on.

Jason2866, 2018-06-30:
Works without any problem. Tested with my easy accessibly sonoff basic "Couch" Modul.
A reading lamp voice controlled via Alexa. So i have not to stand up if i need a light for reading😆

nivek88, 2018-07-01:
Yes, having the same problems!

closed 2018-08-15 by reloxx13