arendst/Tasmota #3103: Interlock on 4ch: off on one port sets all ports off.  (v5.14.0)

chriskmn, 2018-06-30:
Hello,

I am using itead sonoff 4ch pro in interlock mode (setoption14 on). Problem: when I am switching off any port, the active port gets switched off also. E.g. Power1 is set to on, all others off -> the command Power2 off also sets Power1 off. 

Is this an error ?

thanks, Chris

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : itead sonoff 4ch pro
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE

 stat/sonoff-21/STATUS = {"Status":{"Module":23,"FriendlyName":["Sonoff-21","Sonoff-212","Sonoff-213","Sonoff-214"],"Topic":"sonoff-21","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"SaveData":0,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
```

**(Please, remember to close the issue when the problem has been addressed)**

lobocobra, 2018-06-30:
Do you have the latest firmware?
Did you check the physical dip-switches on the ch4pro?
Does the problem happen when you push the buttons, in the web, in the console and mqtt?

Check also your reboot counter... is it rebooting after you switch one off and the other are set off after reboot?

chriskmn, 2018-06-30:
I am pretty sure that the issue is located in the firmware. this is the log from the console:

18:43:42 CMD: setoption14 on
18:43:42 MQT: stat/sonoff-21/RESULT = {"SetOption14":"ON"}
18:43:48 CMD: power2 on
18:43:48 MQT: stat/sonoff-21/RESULT = {"POWER2":"ON"}
18:43:48 MQT: stat/sonoff-21/POWER2 = ON
18:43:56 CMD: power3 off
18:43:56 MQT: stat/sonoff-21/RESULT = {"POWER2":"OFF"}
18:43:56 MQT: stat/sonoff-21/POWER2 = OFF
18:43:56 MQT: stat/sonoff-21/RESULT = {"POWER3":"OFF"}
18:43:56 MQT: stat/sonoff-21/POWER3 = OFF

I switched on "power2" and then switched off power3. The result is switching off power 2 and 3.

lobocobra, 2018-07-02:
Did you check the dip switches on the board?
Did you re-flash the latest firmware to the board?

Without this done, it will not be possible to support you. What I can tell you is, that my CH4pro do not have that same problem. So it can not be a firmware problem. 

!!! CH4pro has a hard wired interlock setting, to be activated with the dip-switch.... either use this one OR the setoption14. Using both will for sure cause problems.

chriskmn, 2018-07-02:
The Wiki says that Tasmota supports also DIP switch mode settings. So there is no reason that a port is going to be switched off due to switching off !!!! any other port. It should only be switched off by a power on command.

DIP switches are as recommended, firmware is reflashed and as stated in the headline it is v5.14.0, what is the latest release.

chriskmn, 2018-07-03:
Meanwhile I tested the v5.14.0 on a 4ch (not pro and without dip switches) and I got the same results. Switching off an inactive port turns off the active port. So it can‘t be an hardware issue.

chriskmn, 2018-07-08:
And if anybody cares, same problem on a third device (4ch) with firmware 6.1.0:

11:54:21 CMD: Setoption14 on
11:54:21 MQT: stat/sonoff-6/RESULT = {"SetOption14":"ON"}
11:54:42 CMD: Power1 on
11:54:42 MQT: stat/sonoff-6/RESULT = {"POWER1":"ON"}
11:54:42 MQT: stat/sonoff-6/POWER1 = ON
11:54:50 CMD: Power2 off
11:54:50 MQT: stat/sonoff-6/RESULT = {"POWER1":"OFF"}
11:54:50 MQT: stat/sonoff-6/POWER1 = OFF
11:54:50 MQT: stat/sonoff-6/RESULT = {"POWER2":"OFF"}
11:54:50 MQT: stat/sonoff-6/POWER2 = OFF

————————————————————————————-
Status 0:
11:58:34 CMD: Status 0
11:58:34 MQT: stat/sonoff-6/STATUS = {"Status":{"Module":7,"FriendlyName":["sonoff-6","sonoff-62","sonoff-63","sonoff-64"],"Topic":"sonoff-6","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":0,"SaveData":0,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
11:58:34 MQT: stat/sonoff-6/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.0.180:81/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T15:57:44","StartupUTC":"2018-07-07T18:00:50","Sleep":250,"BootCount":27,"SaveCount":59,"SaveAddress":"FA000"}}
11:58:34 MQT: stat/sonoff-6/STATUS2 = {"StatusFWR":{"Version":"6.1.0","BuildDateTime":"2018-07-06T21:01:47","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:58:34 MQT: stat/sonoff-6/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"malibu_MQTT","LogPort":514,"SSId":["malibu","malibu"],"TelePeriod":120,"SetOption":["54004008","55818000"]}}
11:58:34 MQT: stat/sonoff-6/STATUS4 = {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","23B6179E","00000000"]}}
11:58:34 MQT: stat/sonoff-6/STATUS5 = {"StatusNET":{"Hostname":"sonoff-6-4284","IPAddress":"192.168.0.20","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"60:01:94:83:B0:BC","Webserver":2,"WifiConfig":4}}
11:58:34 MQT: stat/sonoff-6/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.180","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_83B0BC","MqttUser":"mosquitto","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
11:58:34 MQT: stat/sonoff-6/STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 08 09:58:34 2018","Local":"Sun Jul 08 11:58:34 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":99,"Sunrise":"05:56","Sunset":"21:54"}}
11:58:34 MQT: stat/sonoff-6/STATUS10 = {"StatusSNS":{"Time":"2018-07-08T11:58:34"}}
11:58:34 MQT: stat/sonoff-6/STATUS11 = {"StatusSTS":{"Time":"2018-07-08T11:58:34","Uptime":"0T15:57:44","Vcc":3.142,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"malibu","RSSI":42,"APMac":"54:A0:50:D8:4D:B8"}}}

lobocobra, 2018-07-10:
Well just in case you do care... I have 6 of those devices and none of them has the desribed behavior. Neither on CH4 or CH4 pro.
=> Thus the error has most likely nothing to do with the firmware.
 
So if we assum that your devices are not defective and it is not a firmware issue, then it can only be....
* a problem with the settings you have configured (and yes they remain even if you reflash it)
* a conflict between dip switch settings and your firmware

I know from the past (as I re-programmed the firmware on some of the devices), that the DIP switch settings can interfer. You can see what settings I have on post #824.
Other than that you probably want to flush the settings area and hey.... saying thanks or please, even if a response did not help never did any harm.

Over and out...

closed 2018-07-10 by chriskmn