arendst/Tasmota #3105: Multiple dht22 to domoticz

reydecopas, 2018-06-30:
Sonoff 4ch pro with 3 dht22 works well but it only allows to mqtt one sensor (temp+humidity) to domoticz....

How can I modify the sensor idx type of the others gpio to have them configured as temp+humidity at  the same time?

reydecopas, 2018-07-08:
After updating version I have rules enables  but I get same read for every sensor... what means something is wrong because webpage shows differents values
From Webpage
AM2301-01 Temperature	25.3°C
AM2301-01 Humidity	52.4%
AM2301-02 Temperature	25.0°C
AM2301-02 Humidity	49.6%
AM2301-03 Temperature	25.8°C
AM2301-03 Humidity	48.7%

from console:
status 10
16:12:36 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-07-08T16:12:36","AM2301-01":{"Temperature":25.7,"Humidity":48.6},"AM2301-02":{"Temperature":25.7,"Humidity":48.6},"AM2301-03":{"Temperature":25.7,"Humidity":48.6},"TempUnit":"C"}}

ascillato2, 2018-07-10:
Hi,

The temperature drivers were updated to solve this. Please, retry with the last version in development branch.

Please, reopen this issue if the problem still occur. Thanks.

closed 2018-07-10 by ascillato2

nromensky, 2018-12-18:
interested to know if this feature added? I run sonoff basic with three temp/hum sensors. in order to smoothly integrate with domoticz I need to specify IDX for each dummy TH device, but see only one in sonoff configuration.

arendst, 2018-12-18:
Use rules for the other sensors. See rule examples in wiki.