arendst/Tasmota #3106: ESP8266 Amica

dedynobre, 2018-06-30:
Hello!

I have a esp8266 and I would like to know how to activate the led that is on the board itself.
The mqtt is configured and functioning which is seen to receive the command via console but is unknown.

Jason2866, 2018-06-30:
Hello,

with which GPIO is the LED connected? And what want you to do with the LED?
And what do you mean with "unknow"? Please post some logs to things clearer

dedynobre, 2018-06-30:
The led is connected in GPIO 16 D0.

For test I would like to turn the LED on and off so I can understand the flow of communication

MQT: stat/esp-02/RESULT = {"Command":"Unknown"}

Mqtt configuration:

topic: cmnd/esp-02/D0
value on: 1(on)
value off: 0(off)

Jason2866, 2018-06-30:
Tasmota is in general a software for controlling relay(s).
So you have to connect a (virtual) relay on GPIO 16. This relay can be controlled via webconsole or mqtt.
So take a look at wiki for WemosMini nearly the same device than yours.
https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos%20D1%20Mini

dedynobre, 2018-06-30:
Very good.
Now I can understand.

Thank you.

closed 2018-06-30 by dedynobre