arendst/Tasmota #3108: Somfy RTS Sender Replacement

EddieSch, 2018-06-30:
I looked inside some of the sonoff RF Bridge issues, but could not find a solution for my idea.

I'm using Somfy Telis 4 RTS sender. It works on 433.92 MHz.
I don't want to control the buttons on the sender by a Sonoff device.
I rather want to replace or run in parallel the Sonoff RF Bridge,
to be able to control my electr. shutters.

If one have been able to learn the Sonoff RF Bridge to be able to handle all the Somfy RTS commands,
then there will be a wide range of abilities to integrate existing home devices into smart home solutions
by using Sonoff solutions.

I saw that also the RF chip can be reflashed.

Jason2866, 2018-06-30:
You can sniff the commands. How this work has portisch described
https://github.com/Portisch/RF-Bridge-EFM8BB1
So it is up to you 👍

Jason2866, 2018-06-30:
I have found this.
https://github.com/Nickduino/Somfy_Remote
I think it will not work with the RF Bridge and Tasmota because Somfy has rolling code.

ascillato2, 2018-07-10:
Hi,

Seems that your issue was addressed. Closing it. Please, reopen if needed. Thanks :+1:

closed 2018-07-10 by ascillato2

420Dope, 2018-10-16:
Hi,
i would also control my Somfy shutters with my Sonoff RF Bridge. But i can sniff the commands or something like that because i don´t can programming stuff and other things like that. I only can flash devices with Tasmota.
Are there any news how i can control my Somfy shutters with my Sonoff RF Bridge?

nosaj66au, 2018-11-07:
Hi All can we move this to enhancement now we have RC Remote lib?  This would be a great extension for most China made blinds.

Jason2866, 2018-11-07:
No Somfy isnt compatible to RF Bridge or RC Switch lib.

nosaj66au, 2018-12-11:
Hi @Jason2866  I don't believe that Somfy RTS is rolling code it merely a 24 bit protocol which is not currently supported. I have done a hack up and got it to work mirroring the remote signal. But it would be great as pat of tasmota.

Jason2866, 2018-12-11:
@nosaj66au 
So it isnt Rolling Code at all https://github.com/Nickduino/Somfy_Remote/blob/master/Somfy_Remote.ino
but the code is to complex to integrate in the very hardware limited EFM chip...
So Somfy isnt possible with the RF Bridge. A solution could be if you have the raw data (Bitbucket)
to send it with the Rfraw command. For more Infos see https://github.com/Portisch/RF-Bridge-EFM8BB1

altelch, 2019-01-28:
Wrote a somfy rts decoder. Fill in the EFM8BB1 bucket sniffing B1 line and see the result. Code is located here: https://github.com/altelch/Somfy-RTS

sehaas, 2019-01-31:
I have a Somfy Telis 16 RTS and captured some raw data. Decoding the data with @altelch tool failed, though.
I'm not familiar with this Bitbucket format yet, but I'll try to debug the code.
Does anyone have a valid input string which I could compare to my data?

altelch, 2019-01-31:
Hi,

this is an example input / output I have:

{"RfRaw":{"Data":"AA B1 05 0A14 12FC 04F6 0276 807A 0000000000000012222332223323
33333233222222333323323333332233223323333222332332333323322333322334 55"}}

Manchester Decoded Data-Part

1010 0101 1000 0110 1010111001111011 01100010110011100100010

0xA  0x5  0x8  0x6  0xAE7B 0x62CE44

After deobfuscation

0xA5 0x23 0x28 0xD5 0x19 0xAC 0x8A

Command 0x2 Up

Enter B1 line: {"RfRaw":{"Data":"AA B1 05 0A14 12FC 04F6 0276 7E04 00000000000000122223322332223322233223323333233332233223333332222333323333333333332222333322333334 55"}}
Group       A       B       C       D       F               G                   
Byte:       0H      0L      1H      1L      2       3       4       6       7   
        +-------+-------+-------+-------+-------+-------+-------+-------+-------+
        ! 0xA   + R-KEY ! C M D ! C K S !  Rollingcode  ! Remote Handheld Addr. !
        ! 0xa   +  0x4  !  0x1  !  0x0  !     0x28d4    !        0x19ac8a       !
        +-------+-------+-------+-------+MSB----+----LSB+LSB----+-------+----MSB+

Command 0x1 My / Stop

Rolling code was incremented between my and up, Address is the same. So for me, decoding works.

Could you give me your scanned data and what you pressed?

Greetings,

    Altelch.

sehaas, 2019-01-31:
The bridge receives data during `rfraw 177` sniffing, but I'm not sure if it's only the Somfy remote.
The red LED is responding to my button presses, but does not always generate an output.
Maybe there are also other devices in my neighborhood, which are transmitting.

The tool always fails with `AttributeError: ManchesterDecode instance has no attribute 'bitvec'`. Am I pasting the wrong string?

You can find my data and debug output [here](https://gist.github.com/sehaas/6bf6fe36e9b4d11356f5bcac21bf0fbc).

sehaas, 2019-02-03:
Recording the `Telis 16 RTS` always produces twisted data, but using a `Smoove origin RTS` I got valid codes. I'm now working on a script which generates B0 commands. With this commands I was able to register the RF bridge as own remote-device. So I don't have to always sync the rolling code with the other remotes.
Just one more question:
Are there different codes for single-press-UP (completely open the shades) and holding-UP (move as long as the button is pressed)? I could record and replay only holding-UP.

altelch, 2019-02-04:
No, no different codes but the remote switches to repeat mode and sends 7 times the same frame without incrementing rolling code. I'm also writing a script to create B0 commands but had no success in moving the blind also with the same parameters I could reproduce the exact sequence I sniffed before...maybe you could provide your script...

vaprinz, 2019-02-17:
Why did I always install the product which is not compatible with anything? :-(
That would be so nice if Somfy works.
After all, Somfy has a large Mart share and would make many users happy. Me too, of course.

sory for my english

nosaj66au, 2019-02-18:
Yes, I ended up using blinds from China with the dooya protocol.  Now I need time to integrate it into Tasmota. I have tested the code on an esp which work fine.

sehaas, 2019-02-25:
@vaprinz what kind of remote where you recording? Do you have another type to try with? As I wrote before, I was only able to capture valid B1 strings with one of my remotes.
You can try to switch your shades with in programming mode and register/simulate a new one with this script https://github.com/sehaas/yfmos

GabrielSimoes1983, 2021-07-14:
Sorry to revive this thread but I am trying to figure out how to integrate it using broadlink + HASS.
When I press the butons on my controls, all I can get are B0 messages. How can I generate a B1?

Example of my readings:

b0c00201949c060004390438043804380438043804390439043804370539043a043c043c043b043c043b043d0450043b043b044d043c043e043b043b043c043b0439043a043c043d0438043b043b043a043b043c0449047a043b044c043c043b043e043f04d00437043a043d0438043804730438043a043604460439043a04390439043904390438043a043904390439043804380473043d043c043a043a0439043804380439043804370438043704380438047405390439047304380438043804380439043904380439047205380438043c043e043c0439043a0438043804360439043a044404480439044a04390438043804380439043804390437047a053a00014b0005dc

b0c08402029d06000439043a043804380438043a04750439047404390437043904380438043a043a04380438040d545354539f2a2a2a2a151515152a15151515151515152a2a15152a2a1515151515152a15152a2a2a151515151515151515152a2a2a15152a151515152a15152a151515152a2a2a2a15152a2a1515151515152a2a151515152a00043453545254535452545255525452559e2a2a2a2a151515152a15151515151515152a2a15152a2a1515151515152a15152a2a2a151515151515151515152a2a2a15152a151515152a15152a151515152a2a2a2a15152a2a1515151515152a2a151515152a00043453545255525452555255515453549e2a2a2a2a151516142a15161415151515152a2a15152a2a1515151515152a16142b292a151515151515151515152a2a2a15152a151515152a16142a151515152a2a2a2a1516292a1515151515152a2a151515152a00043552545354525452555255515552549e2b292b29161416152a15151515151514162a2a14162a2a1416141614162916152a292b15151515151515141615292b2916152a151514162a15142b14161515292a2a2b14162a2a141614151516292b141515152a00043553545254525552545254525552589a2b2a2a2a151515152a15151515151515152a2a15152a2a1515141614162a15152a2a2a15151515151515141516292b2a15152a141614162a15152a151515152a2a2a2a15152a2a1515151515152a2a141614162a00043453545254525552555254525552549e2a2a2a2a161416142b14161416141515152a2b1416292b1416141515152a16142a2a2a151515151515151515152a2b2916152a141614152a16142b151415152a2b292b15142b2a1416141614162a2a151415152a0005dc

b0c02202169d060004750438043a04380473043a0438043b043d047a043b04380438043904380419535453539f2a2a2a2a2a151515152a2a151515152a2a2a16142a1515151515152a1614161416142a151614161416292b292a1516292a151629161416152a2a2a2a2a1515151515152a15152a15151515151515151515151515151400042053545255525452555254525552549e2b292b292b15151416292b15151515292b2a15152a1515151515152a1515151515152a1515151515152a2a2a2a15152a2a15152a151515152a2a2a2a2a1515151515152a15152a15151515151515151515151515151400042152545354525552545255525452559e2a2a2a2a2a151515152a2a151515152a2a2a15152a1515151515152a1515151516142a1515151515152a2a2a2b1416292a15152a161416142b2a2a2a2a1515151514162a15152a15151515151515151515151515151400042053545255525452555254525552559d2b292b2a2a15151515292b151515152a2a2a15152a1515151515152a1515151515152a1515151515152a2a2a2a15152a2a15152a151515152b292a2a2b1416141515162916142a16151515151515151514151515161300042153545255525452555255515552559d2b2a2a2a2a151515152a2a151515152a2a2a15152b1416141515152a1515161416142a151614161416292b2a2a14162a2a15152a151515152a2a2a2a2a1515151515152a16142a1515151515151515151515151515140005dc

darenkeates, 2022-09-28:
I am struggling here. I have flashed the rf bridge and updated the rf codes. I can tyoe rfrw 177 into the console and it seems to capture the codes from the remote but when i decode them to b0 messages, it does not work? Is it to do with rolling codes? some people say it does work some people say it doesn't.
Am i doing something wrong?