arendst/Tasmota #3109: Help with sending irsend (coolix)

dindan100, 2018-07-01:
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : NodeMCU
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
08:32:29 CMD: status 0
08:32:29 MQT: stat/ITAYROOM/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"ITAYROOM","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
08:32:29 MQT: stat/ITAYROOM/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T00:03:49","StartupUTC":"2018-07-01T05:28:40","Sleep":0,"BootCount":6,"SaveCount":11,"SaveAddress":"3F9000"}}
08:32:29 MQT: stat/ITAYROOM/STATUS2 = {"StatusFWR":{"Version":"6.0.0b","BuildDateTime":"2018-06-30T14:51:46","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
08:32:29 MQT: stat/ITAYROOM/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SalpeterF",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
08:32:29 MQT: stat/ITAYROOM/STATUS4 = {"StatusMEM":{"ProgramSize":574,"Free":2496,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","00000000","37B6179A","00000000"]}}
08:32:29 MQT: stat/ITAYROOM/STATUS5 = {"StatusNET":{"Hostname":"ITAYROOM-1459","IPAddress":"192.168.1.50","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"2C:3A:E8:17:85:B3","Webserver":2,"WifiConfig":3}}
08:32:29 MQT: stat/ITAYROOM/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.111","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1785B3","MqttUser":"user","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
08:32:29 MQT: stat/ITAYROOM/STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 01 05:32:29 2018","Local":"Sun Jul 01 08:32:29 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":3,"Sunrise":"06:50","Sunset":"22:57"}}
08:32:29 MQT: stat/ITAYROOM/STATUS10 = {"StatusSNS":{"Time":"2018-07-01T08:32:29"}}
08:32:29 MQT: stat/ITAYROOM/STATUS11 = {"StatusSTS":{"Time":"2018-07-01T08:32:29","Uptime":"0T00:03:49","Vcc":2.952,"Wifi":{"AP":1,"SSId":"SalpeterF","RSSI":90,"APMac":"B0:B2:8F:36:C5:EA"}}}

Hi Everyone,

It Might be an simple issue but i cant find docuemntation or exmaple for using the irsend command with coolix enconding....
I have download the latest IRremoteESP8266 library (version is higher than the one provided with the current tasmota zip file) and instaled it on my nodeMCU and was able to send COOLIX encoded mesages to my AC!
So i thought trying using the great Tasmota project to some how connect the nodemcu with the tasmota to my Home assitence and try to use my AC from the HA (still havent got how to do it//// :)

However when i tried from the tasmota console first to send a message like below it seems that that Tasmota dosent know how to use the coolix enconding...
irsend { "Protocol": "COOLIX", "Bits": 24, "Data": 16593103}
08:43:22 MQT: stat/ITAYROOM/RESULT = {"IRSend":"Protocol not supported"}

Am i doing somthing wrong?
should i simply replace the current IRremoteESP8266 in the zip file with the most updated one or i also have to change tasmota files?

Since the current IRremoteESP8266 is working with this enconding and possiably others as well, how do i configure tasmota to undertand it?

Thanks in Advance !!!!

dindan100, 2018-07-02:
Thanks arendst!
Will check the file and see if i can do it myself as i am not a skilled programmer.
In case i will succeed i will update this issue so others will be able to copy and paste.

Thanks

aihysp, 2018-07-08:
yes please leave a guide for us non programmers 👍

closed 2018-09-06 by stale[bot]

errik97, 2019-07-13:
> 
> 
> Thanks arendst!
> Will check the file and see if i can do it myself as i am not a skilled programmer.
> In case i will succeed i will update this issue so others will be able to copy and paste.
> 
> Thanks

Hi, I'm having the same problem. Did you solved? I would like to use coolix protocol with tasmota. Thanks for your help.