arendst/Tasmota #3110: Sonoff Basic goes into reboot loop when emulation is enabled

tlpbu, 2018-07-01:
Thanks you all for your hard work developing Sonoff-Tasmota.        

Today I ran into problem I cannot find any hint in any forum.    
As soon as I enable any type of emulation on Sonoff Basic flashed with Sonoff-Tasmota 5.12 it goes into reboot loop. I have literary few seconds to disable emulation from UI until it reboots again.    
    
I have 4 other Sonoff Basic modules with nearly identical configuration and Belkin Wemo emulation works perfectly on them allowing me to easily control devices from both Amazona Echo and Home Assitant.    
    
Any hint is appreciated.    

*Details about config*
```
21:57:17 CMD: status 0
21:57:17 MQT: stat/sonoff5/STATUS = {"Status":{"Module":1,"FriendlyName":"Bathroom light","Topic":"sonoff5","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":0,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
21:57:17 MQT: stat/sonoff5/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff5.ino.bin","Uptime":"0T00:01:23","Sleep":0,"BootCount":157,"SaveCount":505,"SaveAddress":"FB000"}}
21:57:17 MQT: stat/sonoff5/STATUS2 = {"StatusFWR":{"Version":"5.12.0","BuildDateTime":"2018-04-22T21:45:53","Boot":6,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
21:57:17 MQT: stat/sonoff5/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"6F79EDHo7V1gT65bUgyI","SSId2":"indebuurt2","TelePeriod":300,"SetOption":"00000009"}}
21:57:17 MQT: stat/sonoff5/STATUS4 = {"StatusMEM":{"ProgramSize":493,"Free":508,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
21:57:17 MQT: stat/sonoff5/STATUS5 = {"StatusNET":{"Hostname":"sonoff5-7346","IPAddress":"192.168.0.25","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"***","Webserver":2,"WifiConfig":3}}
21:57:17 MQT: stat/sonoff5/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.2","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_14DCB2","MqttUser":"***","MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:57:17 MQT: stat/sonoff5/STATUS7 = {"StatusTIM":{"UTC":"Sun Jul 01 18:57:17 2018","Local":"Sun Jul 01 21:57:17 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":3}}
21:57:17 MQT: stat/sonoff5/STATUS10 = {"StatusSNS":{"Time":"2018-07-01T21:57:17","Switch1":"OFF"}}
21:57:17 MQT: stat/sonoff5/STATUS11 = {"StatusSTS":{"Time":"2018-07-01T21:57:17","Uptime":"0T00:01:23","Vcc":3.438,"POWER":"OFF","Wifi":{"AP":1,"SSId":"6F79EDHo7V1gT65bUgyI","RSSI":74,"APMac":"50:64:2B:63:34:0A"}}}
```
*This is what I can see on MQTT broker when reboot loop happens*
```
tele/sonoff5/LWT Offline
tele/sonoff5/LWT Online
cmnd/sonoff5/POWER (null)
tele/sonoff5/INFO1 {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_14DCB2","GroupTopic":"sonoffs"}
tele/sonoff5/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff5-7346","IPAddress":"192.168.0.25"}
tele/sonoff5/INFO3 {"RestartReason":"Software/System restart"}
homeassistant/switch/sonoff5/config (null)
stat/sonoff5/RESULT {"POWER":"OFF"}
stat/sonoff5/POWER OFF
tele/sonoff5/STATE {"Time":"2018-07-01T21:54:50","Uptime":"0T00:00:28","Vcc":3.438,"POWER":"OFF","Wifi":{"AP":1,"SSId":"6F79EDHo7V1gT65bUgyI","RSSI":72,"APMac":"50:64:2B:63:34:0A"}}
tele/sonoff5/SENSOR {"Time":"2018-07-01T21:54:50","Switch1":"OFF"}
tele/sonoff5/LWT Offline
tele/sonoff5/LWT Online
cmnd/sonoff5/POWER (null)
tele/sonoff5/INFO1 {"Module":"Sonoff Basic","Version":"5.12.0","FallbackTopic":"DVES_14DCB2","GroupTopic":"sonoffs"}
tele/sonoff5/INFO2 {"WebServerMode":"Admin","Hostname":"sonoff5-7346","IPAddress":"192.168.0.25"}
tele/sonoff5/INFO3 {"RestartReason":"Software/System restart"}
homeassistant/switch/sonoff5/config (null)
stat/sonoff5/RESULT {"POWER":"OFF"}
stat/sonoff5/POWER OFF
tele/sonoff5/STATE {"Time":"2018-07-01T21:55:50","Uptime":"0T00:00:24","Vcc":3.437,"POWER":"OFF","Wifi":{"AP":1,"SSId":"6F79EDHo7V1gT65bUgyI","RSSI":72,"APMac":"50:64:2B:63:34:0A"}}
tele/sonoff5/SENSOR {"Time":"2018-07-01T21:55:50","Switch1":"OFF"}
tele/sonoff5/LWT Online
```

Frogmore42, 2018-07-01:
Look here: https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips

You are using core 2.4.0 which has known issues, especially with Hue emulation, there is a link to a workaround on that page.  I am still using core 2.3.0 mostly, since it seems to be more stable.  I expect someday there will be something newer than 2.4.1 that addresses the current issues.

In the meantime, I don't use Hue Emulation to have Alexa control my stuff, instead I use HA-Bridge (available on GitHub).  It was a bit difficult to set up, but has been very reliable since I started using it.

Jason2866, 2018-07-02:
Hi @Frogmore42 

Alexa works perfect for me with 2.3.0 and V.6.0.0b!
I have disabled "things" i dont use/need. If you are interested i can post later my user_config_override.h

ascillato2, 2018-07-10:
Hi,

Have you managed to solve your issue? If not, please reopen it. Thanks

closed 2018-07-10 by ascillato2

tlpbu, 2018-08-27:
Hi, I uploaded Tasmota 6.1.1 from scratch on that device and fortunatelly haven`t noticed this issue anymore.