arendst/Tasmota #3111: Sonoff Basic slow and losing WiFi at 6.0.0b

nambuco, 2018-07-01:
Sonoff Basic when flashed with 6.0.0b become very unstable, slow and losing WiFi connection from time to time.

Tested in 2 different hardware. Replace firmware to 5.14, both start to be very stable.

In a Sonoff TH I have no problem with 6.0.0b firmware.

Frogmore42, 2018-07-01:
status 0?
This will indicate which core you are using amount other things.

Most likely, you are running into the known and documented issues with the core 2.4.x.
https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips 

And covered in many other issues, this one has a pretty good coverage of the issue
https://github.com/arendst/Sonoff-Tasmota/issues/2689

nambuco, 2018-07-02:
Hello,

I compiled using core 2.3.0 and Visual Code with PlatformIO

Here is the status 0 for 5.14 only because 6.0.0b is too unstable.

5.14.0 status 0

21:40:34 CMD: status 0
21:40:34 MQT: tasmota/sonoff_fermentador/stat/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff"],"Topic":"sonoff_fermentador","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":0,"ButtonRetain":0,"PowerRetain":0}}
21:40:34 MQT: tasmota/sonoff_fermentador/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T01:34:21","StartupUTC":"2018-07-01T23:06:13","Sleep":0,"BootCount":7,"SaveCount":17,"SaveAddress":"FB000"}}
21:40:34 MQT: tasmota/sonoff_fermentador/stat/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-07-01T19:56:03","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
21:40:34 MQT: tasmota/sonoff_fermentador/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["nambuco_home","Time_Machine"],"TelePeriod":60,"SetOption":["04008008","55818000"]}}
21:40:35 MQT: tasmota/sonoff_fermentador/stat/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
21:40:35 MQT: tasmota/sonoff_fermentador/stat/STATUS5 = {"StatusNET":{"Hostname":"sonoff_fermentador-4715","IPAddress":"10.0.1.79","Gateway":"10.0.1.1","Subnetmask":"255.255.255.0","DNSServer":"10.0.1.1","Mac":"2C:3A:E8:07:B2:6B","Webserver":2,"WifiConfig":3}}
21:40:35 MQT: tasmota/sonoff_fermentador/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.0.1.200","MqttPort":1883,"MqttClientMask":"sonoff_fermentador","MqttClient":"sonoff_fermentador","MqttUser":"pi","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
21:40:35 MQT: tasmota/sonoff_fermentador/stat/STATUS7 = {"StatusTIM":{"UTC":"Mon Jul 02 00:40:35 2018","Local":"Sun Jul 01 21:40:35 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":-3,"Sunrise":"00:50","Sunset":"16:57"}}
21:40:35 MQT: tasmota/sonoff_fermentador/stat/STATUS10 = {"StatusSNS":{"Time":"2018-07-01T21:40:35","DS18B20":{"Temperature":5.6},"TempUnit":"C"}}
21:40:35 MQT: tasmota/sonoff_fermentador/stat/STATUS11 = {"StatusSTS":{"Time":"2018-07-01T21:40:35","Uptime":"0T01:34:22","Vcc":3.175,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"nambuco_home","RSSI":24,"APMac":"50:C7:BF:5F:BB:6F"}}}
21:40:47 MQT: tasmota/sonoff_fermentador/tele/STATE = {"Time":"2018-07-01T21:40:47","Uptime":"0T01:34:34","Vcc":3.174,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"nambuco_home","RSSI":26,"APMac":"50:C7:BF:5F:BB:6F"}}
21:40:47 MQT: tasmota/sonoff_fermentador/tele/SENSOR = {"Time":"2018-07-01T21:40:47","DS18B20":{"Temperature":5.5},"TempUnit":"C"}

I will test with core 2.4.1 and report results here.

nambuco, 2018-07-02:
Just updated with 6.0.0b using core 2.4.1. Seems to be stable. I will monitor during the night and close the issue tomorrow if no problems occur.

Thanks.