arendst/Tasmota #3112: Build error: ESP8266SAM.h:22:25: fatal error: AudioOutput.h: No such file or directory

nardev, 2018-07-01:
Just downloaded the repo and directly tried to build it, but i got this error.

I'm wondering if i missed something for platformio?

In file included from .../.platformio/lib/ESP8266SAM_ID2885/src/ESP8266SAM.cpp:20:0:
.../.platformio/lib/ESP8266SAM_ID2885/src/ESP8266SAM.h:22:25: fatal error: AudioOutput.h: No such file or directory

nardev, 2018-07-02:
Direct linking of the lib helped, i'm wondering why pio was not able to find it.

ascillato, 2018-07-02:
Hi

The file you point is not part of Tasmota. Also I don't have it in my platform.io. Seems to be another custom library you install.

So, Try to uninstall and install again.

If you continue to experience this problem with platform.io I recommend you to post the issue in platform.io repo (https://github.com/platformio/platform-espressif8266/issues)

Thanks.

closed 2018-07-02 by ascillato2

nardev, 2018-07-02:
right.. thnx