arendst/Tasmota #3113: Sonoff RF Bridge 433 Not Learning HT6P20B RF Controller

xalexslx, 2018-07-02:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [] Searched the problem in the forum (https://forum.digital-pig.com/)
- [X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : **Sonoff RF Bridge 433 R2 1.0**
- [X] Provide the output of command ``status 0`` :
```
01:33:04 CMD: status 0
01:33:04 SRC: WebConsole from 192.168.0.100
01:33:04 RSL: Group 0, Index 1, Command STATUS, Data 0
01:33:04 MQT: stat/RF-Bridge1/STATUS = {"Status":{"Module":25,"FriendlyName":["RF-Bridge1"],"Topic":"RF-Bridge1","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
01:33:04 MQT: stat/RF-Bridge1/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T01:13:39","StartupUTC":"2018-07-01T23:19:25","Sleep":0,"BootCount":22,"SaveCount":45,"SaveAddress":"F7000"}}
01:33:04 MQT: stat/RF-Bridge1/STATUS2 = {"StatusFWR":{"Version":"6.0.0b","BuildDateTime":"2018-07-01T18:16:04","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
01:33:04 MQT: stat/RF-Bridge1/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["XXXXXX","RF-Bridge1"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
01:33:04 MQT: stat/RF-Bridge1/STATUS4 = {"StatusMEM":{"ProgramSize":541,"Free":460,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","37B6179A","00000000"]}}
01:33:04 MQT: stat/RF-Bridge1/STATUS5 = {"StatusNET":{"Hostname":"RF-Bridge1-0736","IPAddress":"192.168.0.245","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"8.8.8.8","Mac":"XXXXX","Webserver":2,"WifiConfig":3}}
01:33:04 MQT: stat/RF-Bridge1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.140","MqttPort":1883,"MqttClientMask":"RF-Bridge1","MqttClient":"RF-Bridge1","MqttUser":"XXXXXX","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:33:04 MQT: stat/RF-Bridge1/STATUS7 = {"StatusTIM":{"UTC":"Mon Jul 02 00:33:04 2018","Local":"Mon Jul 02 01:33:04 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:51","Sunset":"20:57"}}
01:33:04 MQT: stat/RF-Bridge1/STATUS10 = {"StatusSNS":{"Time":"2018-07-02T01:33:04"}}
01:33:04 MQT: stat/RF-Bridge1/STATUS11 = {"StatusSTS":{"Time":"2018-07-02T01:33:04","Uptime":"0T01:13:39","Vcc":3.207,"Wifi":{"AP":1,"SSId":"XXXXXX","RSSI":100,"APMac":"XXXXXX"}}}
```

Hello,

My Sonoff RF Bridge 433 just arrived and I flashed Tasmota so I can integrate with my HA setup. Also, I flashed the RF Chip firmware according to the Wiki. However, when I try to make it learn my RF controller or motion sensor which uses a HT6P20B chip, it doesn't work. I don't know if I need to learn it or just pass the info to HA via MQTT, but nothing appears on console when I type rfkey1 2 and press a button on my RF controller OR motion sensor.

Jason2866, 2018-07-02:
Known Problem. Stock EFM Firmware doesnt decode this type of code correct.
See #2886

xalexslx, 2018-07-05:
@Jason2866 

Sorry for the delay. The device doesn't output anything after the learning process (only the timed telemetry):
```
04:24:11 CMD: rfkey1 2
04:24:11 MQT: stat/RF-Bridge1/RESULT = {"RfKey1":"Start learning"}
04:24:38 MQT: tele/RF-Bridge1/STATE = {"Time":"2018-07-05T04:24:38","Uptime":"0T00:05:21","Vcc":3.204,"Wifi":{"AP":1,"SSId":"XXXXX","RSSI":100,"APMac":"XXXXXXXXX"}}
```

Here are pics from my two RF devices (Motion sensor and the RF controller):
https://imgur.com/a/Dr2Nd4h

Jason2866, 2018-07-05:
Do you see any output on the web console, when you enter nothing, and just fire your remote or IR Sensor?
There should be a message like RESULT = {RfReceived":{"Sync".............

Jason2866, 2018-07-05:
You can check if your Bridge is working in general with command:
rfraw 1
now when you press your remote you will the raw RF Data.
In this case the protocol your equipment use is not supported.
Take a look here how you can sniff and implement...
https://github.com/Portisch/RF-Bridge-EFM8BB1

Jason2866, 2018-07-05:
Code is not supported

http://support.iteadstudio.com/support/discussions/topics/11000015088/page/1?sort=popularity&url_locale=

Only way is try to implement over RAW or use other hardware for PIR

Jason2866, 2018-07-05:
Rfraw could show something like this
17:27:33 CMD: rfraw 1
17:27:33 MQT: stat/sonoff-8E8329/RESULT = {"RfRaw":"ON"}
17:27:40 MQT: tele/sonoff-8E8329/RESULT = {"RfRaw":{"Data":"AAA42C100172044C15055455"}}
17:27:40 MQT: tele/sonoff-8E8329/RESULT = {"RfRaw":{"Data":"AAA42BFC017C044215055F55"}}
17:27:40 MQT: tele/sonoff-8E8329/RESULT = {"RfRaw":{"Data":"AAA42C06017C043815155F55"}}
If it shows nothing try rfraw 177
17:29:05 CMD: Rfraw 177
17:29:05 MQT: stat/sonoff-8E8329/RESULT = {"RfRaw":"ON"}
17:29:05 MQT: tele/sonoff-8E8329/RESULT = {"RfRaw":{"Data":"AAA055"}}
17:29:10 MQT: tele/sonoff-8E8329/RESULT = {"RfRaw":{"Data":"AA B1 04 01CA 047A 011B 2BFC 01010110011001100101010101100110011001100110010103 55"}}
17:29:10 MQT: tele/sonoff-8E8329/RESULT = {"RfRaw":{"Data":"AA B1 03 0170 0441 2BFC 01010110011001100101010101100110011001101010101002 55"}}
17:29:11 MQT: tele/sonoff-8E8329/RESULT = {"RfRaw":{"Data":"AA B1 04 01D3 0478 0116 2BFC 01010110211021102121212101100110011001100110010103 55"}}
17:29:11 MQT: tele/sonoff-8E8329/RESULT = {"RfRaw":{"Data":"AA B1 03 016F 0441 2BFC 01010110011001100101011001100110011001101010101002 55"}}

xalexslx, 2018-07-05:
I used rfRaw 177 and I got an output like this

```
05:21:18 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 01FD 03DF 0082 2CB0 0100101010101011010100101001010100110100110101001011001100 55"}}
05:21:27 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 01EC 03E3 2CA6 0100101010101011010100101101010100110100110101001011001100 55"}}
05:21:28 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 05 01FA 03DB 006E 0140 2CB0 010100101010101011010100101101010100110100110101001011001100 55"}}
05:21:44 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 03D8 01FF 041A 2CB0 1011010101010100101011010010101011001011001010110100110010 55"}}
05:21:45 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 07 01FE 03D9 1621 007F 019A 00FA 2CB0 0100101010101011010100101101010100110100110101001011001100 55"}}
05:21:46 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 01F9 03DF 2CBA 0100101010101011010100101101010100110100110101001011001100 55"}}
05:21:50 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 03E3 01F1 006E 2C42 1011010101010101001011010010101011001011001010110100110010 55"}}
05:21:51 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 04 01FD 03DA 006E 29D6 0100101010101011010100101001010100110100110101001011001100 55"}}
05:21:52 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 05 01F1 03E3 0140 00DC 2CA6 0100101010101011010100101101010100110100110101001011001100 55"}}
05:21:54 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 01FC 03DE 2CBA 0100101010101011010100101101010100110100110101001011001100 55"}}
05:21:54 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 03 0219 03DD 2CBA 0100101010101011010100101101010100110100110101001011001100 55"}}
05:21:57 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 05 0226 03E6 09D8 0104 292C 0100101010101011010100101101010100110100110101001011001100 55"}}
05:22:00 MQT: tele/RF-Bridge1/RESULT = {"RfRaw":{"Data":"AA B1 06 03DD 0407 01FB 017C 00B4 2CA6 2022020202020200202022020020202022002022002020220200220020 55"}}
```

According to this link (https://acturcato.wordpress.com/2014/01/04/decoder-for-ht6p20b-encoder-on-arduino-board-english/) it is possible to learn this protocol using a normal receiver on arduino. I believe if this is possible, can we make it work on RF Bridge as well?

On the same pages it describes how the protocol is:

- The HT6P20B generates a code 28-bit, 22-bit for ADDRESS, 2-bit for DATA and 4-for “ANTI-CODE” (verification code).
- The ADDRES is fixed for a given Encoder (each HT6P20 has a different address).
- The DATA changes according to the pressed button.
- ANTI-CODE is always the same: “0101”.
- For HT6P20D model there are 20-bit for ADDRESS and 4-bit for DATA.
- For HT6P20F model there are 19-bit for ADDRESS and 5-bit for DATA.

There is something else I can do? I'm probably not skilled to implement it myself.

Jason2866, 2018-07-05:
At the moment i try to understand and learn how sniffed raw code (with command 177) successfull can be sended -> #3124

mr-sneezy, 2018-07-06:
RFRAW 177 changes the result, I now get some response from the console for this device. The crazy thing is, I've been using this device for six months with Home Assistant and the same RF Bridge with no issues BEFORE upgrading to the Portisch firmware !

This is what I get with a long press of the door bell.
`09:41:06 RSL: Group 0, Index 1, Command RFRAW, Data 177
09:41:06 MQT: stat/sonoff_bridge/RESULT = {"RfRaw":"ON"}
09:41:06 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AAA055"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00A0 01EA 0082 13BA 01011010101001101010011001010110010101010110011003 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 04 00C3 01E2 0080 13EC 01011010101021101010211021210110212121010110011023 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 009E 01EA 13C4 01011010101001101010011001010110010101010110011002 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A2 01EA 13C4 01011010101001101010011001010110010101010110011002 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A7 01E7 13BA 01011010101001101010011001010110010101010110011002 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A1 01EB 13C4 01011010101001101010011001010110010101010110011002 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A3 01EF 13BA 01011010101001101010011001010110010101010110011002 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A5 01EA 13BA 01011010101001101010011001010110010101010110011002 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 01EF 13C4 01011010101001101010011001010110010101010110011002 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A1 01EC 13C4 01011010101001101010011001010110010101010110011002 55"}}
09:41:13 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A1 01EA 13BA 01011010101001101010011001010110010101010110011002 55"}}
09:41:14 MQT: tele/sonoff_bridge/RESULT = {"RfRaw":{"Data":"AA B1 03 00A0 01EA 13C4 01011010101001101010011001010110010101010110011002 55"}}`

xalexslx, 2018-07-06:
@mr-sneezy 

I think your case is different because my device didn't worked before Portisch as well. Therefore, I just think I need to know how to make the firmware learn the HT6P20B protocol.

mr-sneezy, 2018-07-07:
Yes I'm thinking the same. I should make a new issue thread.

ChristianKrausse, 2018-07-25:
@xalexslx I have two remotes, with both I am unable to learn with  "RfKey x 2". One Remote (Brennenstuhl RCT CE1 CR3726) gave no output on consol while trying to learn. The other one (KWMOBILE 43965 Freq. Code 0304 [label on the remote]) seem to gave some input while learnig phase, but if I try to send the code with RfKey x 1 (RfKey1 1) it seems to want to learn again and no reaction on the recieving output...

Is there anyone who could explain how to use the always altering Output which I could capture with 

"rfRaw 177"
Always pressing "ON"  at the Remote
`22:54:37 CMD: rfRaw 177 
22:54:37 MQT: stat/Sonoff-RF433-Bridge/RESULT = {"RfRaw":"ON"}
22:54:42 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C021203F2AD1F5C55"}}
22:54:43 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C020803FCAD1F5C55"}}
22:54:48 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C48020803FCA95BAC55"}}
22:54:51 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C52020803F2AD1F5C55"}}
22:54:54 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01FE03FCA5A76C55"}}
22:54:56 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C021C03DEAFFE1C55"}}
22:54:58 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C020803F2AD1F5C55"}}
22:55:02 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C52020803F2A95BAC55"}}
22:55:02 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C52020803F2A95BAC55"}}
22:55:04 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5201FE03FCA5A76C55"}}
22:55:06 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C52021203E8AFFE1C55"}}
22:55:08 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C52020803F2AD1F5C55"}}
22:55:10 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5201FE03FCA95BAC55"}}
22:55:12 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01FE03FCA5A76C55"}}
22:55:15 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C021C03DEAFFE1C55"}}
22:55:17 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C020803F2AD1F5C55"}}
22:55:20 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5201FE03FCA95BAC55"}}
22:55:22 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C52020803F2A5A76C55"}}
22:55:24 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C021203E8AFFE1C55"}}`

Always pressing Off on the remote:
`22:57:04 CMD: rfRaw 177 
22:57:05 MQT: stat/Sonoff-RF433-Bridge/RESULT = {"RfRaw":"ON"}
22:57:10 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01F40410A3322C55"}}
22:57:14 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01F40410AB068C55"}}
22:57:15 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C84020803FCA07DBC55"}}
22:57:18 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C7001FE0406AC947C55"}}
22:57:20 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01F40410A3322C55"}}
22:57:22 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C7001F40410AB068C55"}}
22:57:23 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01F40410A3322C55"}}
22:57:24 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C66020803FCA07DBC55"}}
22:57:25 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5201F40410A3322C55"}}
22:57:26 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01F40410AB068C55"}}
22:57:27 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C6601FE0406AC947C55"}}
22:57:29 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01F40410A3322C55"}}
22:57:30 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01F40410AB068C55"}}
22:57:31 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C70020803FCA07DBC55"}}
22:57:32 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C6601FE0406AC947C55"}}
22:57:33 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01F40410A3322C55"}}
22:57:33 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01F40410AB068C55"}}
22:57:34 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C70020803FCA07DBC55"}}
22:57:35 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C5C01FE0406AC947C55"}}
22:57:35 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C6601FE0406AC947C55"}}
22:57:36 MQT: tele/Sonoff-RF433-Bridge/RESULT = {"RfRaw":{"Data":"AAA41C6601F40410A3322C55"}}`

ChristianKrausse, 2018-07-25:
Also could someone expain how to use the Line:
RfRaw | 169 or AAA955 | (Portisch) Start learning predefined protocols
from https://github.com/arendst/Sonoff-Tasmota/wiki/Commands ? If I do this  I get output like the one from my last comment. But how do I use the maybe learned stuff? And is there some stop command? I curse myself for not buying again the old RF-Poweroutlets with DIP-Switches.

xalexslx, 2018-08-05:
@ChristianKrausse 

This might help you:
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/22

I didn't try to work with RF Bridge more for the time being as I'm focusing on another part of my automation.

ascillato2, 2018-08-25:
Hi,

Closing issue as in the referenced issue it said that was solved.

Please, ask to reopen if needed.

Thanks

closed 2018-08-25 by ascillato2