arendst/Tasmota #3115: Problem running LED Lights and Motion Sensor on a single WeMos D1 Mini

dimaj, 2018-07-02:
Hey guys,

I was hoping someone could help me out with a small project... I am running a WeMos D1 mini and I am trying to connect 2 devices to it: 
 * LED lights ([WS2812B](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-WS2812B-RGB-Shield)) 
 * Microwave Radar Sensor ([RCWL-0516](https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-RCWL-0516-Microwave-Radar-Sensor)). 

I've also followed both guides on how to set up both devices (links above), which led me to this configuration:
* D1 = Data pin for the radar sensor
* D2 = Data pin for the LED lights

The problem that I am facing is that there's a _Toggle_ button in the UI that is being controlled by both radar sensor and led lights, which results in constant flickering.

Is there a way for me to have both devices connected to the same wemos board but to have toggle button assigned just to the led lights? Ideally, I'd like to be notified of motion and then toggle my lights via OpenHAB or Home Assistant.

Thanks for the help and great software!

Frogmore42, 2018-07-03:
Take a look at the wiki here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

You can set up Switch Topic so that it does not actually switch the light.

Or, you can configure it with two things (both switch and relay, but the 2nd relay doesn't need to be hooked up to anything).

dimaj, 2018-07-03:
Thanks for the reply!

I have looked at it before, but didn't think it'll apply to me. 

In any case, I've followed your advice and I no longer have flickering lights. So, Thank you! To get to this stage, I have executed this command:
```
SwitchTopic motion-sensor
```

Now, when there's motion, I get the following in my console logs:
```
cmnd/motion-sensor/POWER1: TOGGLE
```

So, that brings up 2 questions:
1) Why is payload `TOGGLE` instead of `ON`/`OFF` and
2) Is there any way to make it report `ON`/`OFF`?

Thanks again for the help.

Frogmore42, 2018-07-03:
Go read the part about SwitchMode, one of them might do more what you want (mode 1 or 2 is probably closer to what you want).

dimaj, 2018-07-04:
got it to work just the way I wanted to! Thanks for the help!

In case someone is struggling, like I was, here's what I did:
Connect LED light strip to D2
Connect Motion Sensor to D1
Went to console an typed in:
```
SwtichTopic motion
SwitchMode 2
```

After that, I was able to handle my light switching from home assistant via 2 rules.

closed 2018-07-04 by dimaj