arendst/Tasmota #3116: Unable to set/remove MQTT Username/Password

NOTORIOUSVR, 2018-07-02:
It seems I am unable to change the MQTT user and pass from the Web GUI when there is no host defined.  If I set a host and then attempted to change (ie. remove as my MQTT server doesn't have a login) the user and pass the Web GUI crashes and doesn't recover until I reboot physically 1-2 times.

- [ ] Hardware used :  SONOFF S31
- [ ] Provide the output of command ``status 0`` :
```
23:34:05 RSL: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["PWR02"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
23:34:05 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:03:52","StartupUTC":"2018-07-02T22:30:13","Sleep":0,"BootCount":25,"SaveCount":35,"SaveAddress":"F9000"}}
23:34:05 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-07-02T17:21:06","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
23:34:05 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["BATCAVE",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
23:34:05 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":545,"Free":456,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","17B6179A","00000000"]}}
23:34:05 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"PWR02","IPAddress":"10.10.10.124","Gateway":"10.10.10.254","Subnetmask":"255.255.240.0","DNSServer":"10.10.10.10","Mac":"DC:4F:22:0A:EE:D9","Webserver":2,"WifiConfig":3}}
23:34:05 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0AEED9","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
23:34:05 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Jul 02 22:34:05 2018","Local":"Mon Jul 02 23:34:05 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:51","Sunset":"20:57"}}
23:34:05 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-07-02T23:34:05"}}
23:34:05 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-07-02T23:34:05","Uptime":"0T00:03:52","Vcc":3.523,"POWER":"OFF","Wifi":{"AP":1,"SSId":"BATCAVE","RSSI":100,"APMac":"80:2A:A8:81:E9:3F"}}}
```

**(Please, remember to close the issue when the problem has been addressed)**

MYLE-01, 2018-07-03:
have you tryed sending MqttPassword 0 in the console and MqttHost 0

NOTORIOUSVR, 2018-07-03:
MqttPassword 0 worked....   so then I thought to try MqttUser 0 and after sending that command the Web GUI immediately became unresponsive shortly and then reappeared...   the problem with these commands is that they're not clearing the user/pass, they're setting them to "0" but I need them to be blank which doesn't seem possible. 

Deleting the password field in the GUI and saving it is immediately repopulated after the restart with some defaults.  Is there an area in the sketch that has MQTT user/pass set that I can attempt to remove there?  

If yes, is there a way to push the firmware locally without resoldering headers and flashing via USB?

reloxx13, 2018-07-04:
>  the problem with these commands is that they're not clearing the user/pass, they're setting them to "0" but I need them to be blank which doesn't seem possible.

this is not correct.

wiki:
it sets the username/pw to none, means the mqtt connection does not use a username/pw to connect.

try tasmotamqtt in user_config.h and not the default mqtt library.

NOTORIOUSVR, 2018-07-05:
Oddly enough, with all this playing around I had noticed the topic was actually causing the issue... it was set to sonoff, and changing it to something unique seemed to have fixed the issues I was having and the device and web server is working flawless.  

Re. the MqttUser/Pass 0 command, I only thought it was the case as after the console commands going to the MQTT confiuration page you see the user field has a 0 in it, and the password field just one character as well.

Either case, it seems setting a proper/unique topic is the key here to avoid issues I was seeing.

Thanks again!

closed 2018-07-05 by NOTORIOUSVR

dachshund-digital, 2020-11-21:
I am trying to send MQTT power off command to my smart plug, with NO USER and NO PASSWORD for MQTT broker... and it is NOT? possible firmware 8.3.1?  Every time the plug restarts, the default user and password is present.  How do I resolve this?