arendst/Tasmota #3117: Reset

detsadrb, 2018-07-02:
Does anybody can develop feature like this:

when power consumption more or Less then input value — sonoff switch off or switch off (delay 3 sec) and turn on. It will amazing if i could to do hardware reset with sonoff pow.

Frogmore42, 2018-07-03:
You can easily do this using NodeRED and probably other home automation systems too.  I do something even more complicated to control my dehumidifier.

Frogmore42, 2018-07-04:
If this answers your question, please close the issue.

curzon01, 2018-07-05:
Without an external system use [Rules](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules) with a [Backlog command](https://github.com/arendst/Sonoff-Tasmota/wiki/Rules#command)

closed 2018-07-10 by ascillato2