arendst/Tasmota #3121: Use Sonoff RF Bridge's internal buzzer via MQTT commands ?

mr-sneezy, 2018-07-03:
The Sonoff RF Bridge uses an internal buzzer during code learning for indicating the start and stop of that process. 
Is it possible to also use this buzzer as an indicator device during normal operation via MQTT from home automation ?
The purpose could be for a door chime, a general warning device, anything at all that automation monitors.
This would save making a separate MQTT device for this purpose.

arendst, 2018-07-03:
Nope. Buzzer is controlled by second microcontroller.

mr-sneezy, 2018-07-03:
Oh, thanks, yes that would kill that idea.

Jason2866, 2018-07-03:
Why not connect a extra cheap buzzer on a free GPIO Pin? There is enough room for it in the bridge
Defined as relay, so you can switch it on and off via mqtt.

mr-sneezy, 2018-07-03:
Yep that sounds OK. GPIO 4 & 5  are used for programming the EFM8 I think (correct me if I'm mistaken) so it would need to be GPIO 2 ?

Jason2866, 2018-07-03:
Yes GPIO 2 is left free. If you dont flash your EFM you can use GPIO 4 and 5 too.
Only used during EFM flashing process!

mr-sneezy, 2018-07-03:
I did the permanent wiring mod for the EFM, so it will need to be GPIO2. I think my small buzzers are 5V so I'll fit a 2N7000 buffer.

ErikAndren, 2018-07-04:
If you use the portisch RF firmware I do belive there is a UART command to beep the buzzer.
I'm unsure if this can be triggered out of the box via an MQTT message otherwise some minor code modification of the main tasmota firmware might be needed.

mr-sneezy, 2018-07-04:
That's interesting if I understand. You mean the main MCU subscribes to a topic (maybe called 'sonoff_bridge/buzzer') and it sends a serial command to the EFM chip to 'honk it's horn' so to speak when it gets the topic condition 'on'  ?
Would the EFM be able to do that and not drop detection of RF devices while doing it ?

ErikAndren, 2018-07-04:
It won't drop devices but the EFM8bb1 software is not multi threaded i. e. the efm8bb1 will be blind to RF messages during the time while it is beeping. 
This could likely also be improved in the efm8bb1 software but for now it does nothing while waiting for the beep to happen.

arendst, 2018-07-04:
I forgot but just tested. With the portisch update as available in the development branch you can use command ``rfraw 192`` or rfraw aac002c055`` to sound the beep. So ``cmnd/nridge2/rfraw aac002c055`` does the job.

ascillato2, 2018-07-10:
Hi,

Your question seems to be addressed. Closing this issue. Please, reopen if needed. Thanks.

closed 2018-07-10 by ascillato2

dibblerbt, 2021-05-21:
Standard tasmota buzzer command with an easy hardware hack: 
Connect with simple resistor (between 100ohm and 500ohm) from Gpio4 (IO4 marked) to R16 solder-pad which is closest to the buzzer (marked X below, this pad goes to gate of Q2).
Change config IO4 to buzzer.
Voila, ready. 
If 100Ohm=loud, 500Ohm=silent. I found at home 220ohm, moderately loud. This is a 2n7002 enhanced mode FET.
Disadvantage: after this hack the efm controller pio beep will be really silent, but with tasmota we don't use this beep. 

O  O
R   R
1   1
6   7
X   O