arendst/Tasmota #3122: Status when as a result of MQTT command doesn't use configured topic.

landrews33, 2018-07-03:
FullTopic is defined as /%topic%/%prefix%/

If I toggle via the UI or Console - the result is published to the expected MQTT topic.
16:17:56 MQT: /shedlight/stat/RESULT = {"POWER1":"ON"}
16:17:56 MQT: /shedlight/stat/POWER1 = ON (retained)
If I toggle via MQTT - the topic doesn't match the config.
16:18:08 MQT: stat/shedlight/RESULT = {"POWER1":"ON"}
16:18:08 MQT: stat/shedlight/POWER1 = ON (retained)

Jason2866, 2018-07-03:
Something is wrong with your topic... you have a / at the beginnig. That is strange.
It has too look like this
`17:33:56 MQT: tele/sonoff-065CA6/STATE = {"Time":"2018-07-03T17:33:56","Uptime":"3T05:50:33","Vcc":3.122,"POWER":"OFF","Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","RSSI":100,"APMac":"00:A0:57:1E:69:50"}}
17:36:33 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"ON"}
17:36:33 MQT: stat/sonoff-065CA6/POWER = ON
17:36:35 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"OFF"}
17:36:35 MQT: stat/sonoff-065CA6/POWER = OFF
17:36:37 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"ON"}
17:36:37 MQT: stat/sonoff-065CA6/POWER = ON
17:36:39 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"OFF"}
17:36:39 MQT: stat/sonoff-065CA6/POWER = OFF
17:36:43 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"ON"}
17:36:43 MQT: stat/sonoff-065CA6/POWER = ON
17:36:45 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"OFF"}
17:36:45 MQT: stat/sonoff-065CA6/POWER = OFF
17:36:47 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"ON"}
17:36:47 MQT: stat/sonoff-065CA6/POWER = ON
17:36:48 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"OFF"}
17:36:48 MQT: stat/sonoff-065CA6/POWER = OFF
17:37:08 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"ON"}
17:37:08 MQT: stat/sonoff-065CA6/POWER = ON
17:37:09 MQT: stat/sonoff-065CA6/RESULT = {"POWER":"OFF"}
17:37:09 MQT: stat/sonoff-065CA6/POWER = OFF`

landrews33, 2018-07-03:
The / is due to the setup of the broker (it prepends the topic)

Seems this is because of

 fallback_topic_flag = (strstr(topicBuf, mqtt_client) != NULL);

If mqtt_client is a substring of the topic - the goes into fallback.   Suggest this is documented as I cannot find any reference to this.

ascillato2, 2018-07-10:
Hi,

If your question is addessed, please close this issue. Thanks :+1:

ascillato2, 2018-07-16:
Hi

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2