arendst/Tasmota #3123: Question: does anyone have implemented a wind speed meter (anemometer)?

chriskmn, 2018-07-04:
I am wondering if there is some interest in implementing a wind speed sensor like e.g. the Somfy SM 9127932. It is available for less then 20€ but I don‘t know anything about the interface.

I would by one and investigate on the hardware but then I need some support on the firmware.

Any comments on this ?

Jason2866, 2018-07-04:
Hi,

i dont think that the Somfy is a exact anemometer. If i understand the manual of the system correct it is a simple adjusted sensor to switch.

[image]

chriskmn, 2018-07-04:
Hi,

yes I was also thinking about that. Well, I‘ll give it a try and see how the electronic do look like. May be I could get a PWM out of it or so.

closed 2018-07-13 by chriskmn