arendst/Tasmota #3124: Sonoff RF Bridge Controlling PowerGlide Blinds (hpr433a)

angustyler, 2018-07-04:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff RF Bridge V1 running Tasmota 6.0.0a and Portisch firmware on EFM8BB1
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
14:03:31 CMD: status 0
14:03:31 MQT: stat/sonoff_6.0.0a/STATUS = {"Status":{"Module":25,"FriendlyName":["sonoffbridge"],"Topic":"sonoff_6.0.0a","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
14:03:31 MQT: stat/sonoff_6.0.0a/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:00:24","StartupUTC":"2018-07-04T13:03:07","Sleep":0,"BootCount":24,"SaveCount":31,"SaveAddress":"F9000"}}
14:03:31 MQT: stat/sonoff_6.0.0a/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-25T16:51:45","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
14:03:31 MQT: stat/sonoff_6.0.0a/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["SpeedTouch1F866F",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
14:03:31 MQT: stat/sonoff_6.0.0a/STATUS4 = {"StatusMEM":{"ProgramSize":572,"Free":428,"Heap":11,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","37B6179A","00000000"]}}
14:03:31 MQT: stat/sonoff_6.0.0a/STATUS5 = {"StatusNET":{"Hostname":"sonoff_6.0.0a-4711","IPAddress":"192.168.xx.xx","Gateway":"192.168.xx.xx","Subnetmask":"255.255.255.0","DNSServer":"192.168.xx.xx","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":2}}
14:03:31 MQT: stat/sonoff_6.0.0a/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xx.xx","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_xxxxxx","MqttUser":"xxxxxxx","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
14:03:31 MQT: stat/sonoff_6.0.0a/STATUS7 = {"StatusTIM":{"UTC":"Wed Jul 04 13:03:31 2018","Local":"Wed Jul 04 14:03:31 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:51","Sunset":"21:17"}}
```
Please let me know if I have posted this in the wrong place and I will close/move it.

With Tasmota and Portisch installed on my Sonoff RF Bridge, I’ve been trying to automate my PowerGlide blinds but get no response from them. The remote for the blinds has an HPR433A RF unit.
By pressing the button multiple times, I captured the following data from the web console:
AA	B1	05	1572	0654	02D1	015E	4104	01233223322332323232233232323232323230	55
AA	B1	05	152C	0654	02D3	0172	40F0	01233223322332323232233232323232323230	55
AA	B1	05	1536	0654	02CC	016C	41FE	0123322332233232323223323232323232323230	55
AA	B1	05	1522	065E	027B	0181	3E20	0123322332233232323223323232323232323230	55
AA	B1	05	1536	064A	0289	0186	3E02	0123322332233232323223323232323232323220	55
AA	B1	05	152C	0654	02D7	0181	415E	0123322332233232323223323232323232323230	55
AA	B1	05	152C	0654	0280	017F	3EDE	0123322332233232323223323232323232323220	55
AA	B1	05	152C	0654	02D8	0195	458C	0123322332233232323223323232323232323230	55
AA	B1	05	152C	0654	02A1	0184	3D94	0123322332233232323223323232323232323220	55
AA	B1	05	152C	064A	02D8	018E	3FF6	0123322332233232323223323232323232323230	55
AA	B1	05	152C	0654	029D	0186	3D58	0123322332233232323223323232323232323220	55
AA	B1	05	1536	064A	02DA	0192	3FEC	0123322332233232323223323232323232323230	55
AA	B1	06	152C	064A	02D9	017B	010E	3BA0	0123322332233232323223323232323232323220	55
AA	B1	05	152C	065E	02DF	014F	3C7C	0123322332233232323223323232323232323220	55
AA	B1	05	152C	065E	028B	0172	3E20	0123322332233232323223323232323232323220	55
AA	B1	05	152C	0654	02DB	0156	3BA0	0123322332233232323223323232323232323220	55
AA	B1	05	152C	065E	02DA	0163	3A4C	0123322332233232323223323232323232323220	55
AA	B1	05	152C	0654	02DB	0173	3DF8	012332233223323232322332323232323232322330	55
AA	B1	05	1522	065E	02D9	01A5	37CC	01233223322332323232233232323232323232233230	55
AA	B1	05	1522	065E	02E2	0156	3AD8	012332233223323232322332323232323232322330	55
AA	B1	05	1522	0654	028F	0177	368C	01233223322332323232233232323232323232233230	55
AA	B1	05	1518	065E	028F	0174	3678	01233223322332323232233232323232323232233230	55
AA	B1	05	1522	0654	02DF	0174	3BFA	01233223322332323232233232323232323232233220	55
AA	B1	06	1522	065E	02DD	0174	00F0	36A0	01233223322332323232233232323232323232233220	55
AA	B1	06	1522	065E	02E0	0171	00F0	36D2	01233223322332323232233232323232323232233220	55

These lines appeared on the console in the following format:
19:20:16 MQT: tele/sonoff_6.0.0a/RESULT = {"RfRaw":{"Data":"AA B1 05 152C 0654 02DB 0156 3BA0 0123322332233232323223323232323232323220 55"}}

Ignoring the rows with 6 buckets of timing data, I tried:

Field 1: User AA to mark UART Sync Init
Field 2: Replaced the B1 with B0 transmit bucketed RF data
Field 3: Put 28 (i.e. 40 decimal) – guessing this is the number of digits in the RF data to be transmitted
Field 4: Used “05” as this is the number of timing buckets
Field 5: Used “01” as assume repeating it once should be enough (noting original remote only transmits once)
Field 6-10: The average value for each of the five columns of timing data
Field 11: Took the most common value set for the RF data to be sent
Field 12: used 55 to mark UART End Sync
AA	B0	28	05	01	152C	0656	02C1	0177	3DDD 0123322332233232323223323232323232323220	55

Hence, I executed the following on the console:
rfraw 0xAAB0280501152C065602C101773DDD012332233223323232322332323232323232322055
Which flashed the red transmit LED and returned the result:
10:10:13 MQT: stat/sonoff_6.0.0a/RESULT = {"RfRaw":"ON"}
10:10:03 MQT: tele/sonoff_6.0.0a/RESULT = {"RfRaw":{"Data":"AAA055"}}
and no response from the blinds. I tried many times from very close to a few meters away with this same result every time.

I have also tried the same process to the Sonoff remote keyfob to drive a Sonoff RF switch. The keyfox successfully toggles the switch and I can capture the codes. However, when IO transmit the codes, the Sonoff RF switch does not react. I tried this with 0xA6/A5 and 0xB1/B0 commands, both of which appeared to give usable data but with no reaction from the switch.

Any help hugely appreciated as I have run out of ideas!

Jason2866, 2018-07-05:
Do you have tried to learn the Sonoff remote keyfob to the bridge and let the bridge send this learned code? Does the Sonoff RF switch? If yes take a look at the raw code. Try to send this raw code to Sonoff RF. Have you success switching the RF Sonoff with the sended RAW code?

Jason2866, 2018-07-05:
I would try increasing repeating the sended code "Field 5". In Portischs example it is 26!

Jason2866, 2018-07-05:
For me Field 3 is a miracle! Who to get the correct value? The example on Portisch side didnt helped me either...

> Bucket Transmitting using command 0xB0
This command accommodates RF protocols that can have variable bit times. With this command, up to 16 time buckets can be defined, that denote the length of a high (mark) or low (space) transmission phase, e.g. for tri-state mode RF protocols. This command also accommodates code repetition often used for higher reliability.

Hex: AA B0 20 04 1A 0120 01C0 0300 2710 01212122012201212121212121220121212201212203 55

0xAA: uart sync init
0xB0: transmit bucketed RF data
0x20: data len: 32 bytes
0x04: number of buckets: 4
0x19: number of repetitions: (transmit 1+25 = 26 times)
0x01-0x20: Bucket 1 length: 288µs
0x01-0xC0: Bucket 2 length: 448µs
0x03-0x00: Bucket 3 length: 768µs
0x27-0x10: Bucket 4 length: 10ms (sync)
0x05-0xDC: SYNC_LOW
0x02-0xBC: BIT_HIGH_TIME
0x01-0x03: RF data to send (high/low nibbles denote buckets to use for RF high (on) and low (off))
0x55: uart sync end

angustyler, 2018-07-07:
Jason, thank you for your suggestions. 
I'm away for the next 2-3 weeks. Will try more tests with the Sonoff key fob and with variable repetitions on my return.

angustyler, 2018-07-26:
Now that I've been back a few days, I've had time to play:
1. I increased the repeat transmit value progressively to the point where it was 0x20 which resulted in several seconds of transmission (i.e. red led flashing continuously) but no reaction from the blinds.
2. I've started again with the keyfob. No success yet but still trying. I figure this should be a more simple and known quantity so will work on this.
Am beginning to wonder if it is me doing something wrong, or whether my RF Bridge is not working properly. I might try using a Broadlink to sniff the data from the key fob and the data from the RF Bridge to see if there is a difference.
Will provide a more detailed update shortly...

Jason2866, 2018-07-26:
Maybe this informations helps you
https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/22

ascillato2, 2018-08-25:
Hi,

Closing issue as in the referenced issue it said that was solved.

Please, ask to reopen if needed.

Thanks

closed 2018-08-25 by ascillato2