arendst/Tasmota #3125: Store "Total, Yesterday, Today and Period" telemetry values into persistent storage

andreibanaru, 2018-07-04:
**Have you look for this feature in other issues and in the wiki?**
Yep

**Is your feature request related to a problem? Please describe.**
If a Sonoff Pow device restart the counters for today_kwh and others will be reset to 0.

**Describe the solution you'd like**
Store the counters into the persistent storage.

**Describe alternatives you've considered**
N/A

**Additional context**
For example below between 8 AM and 12 PM I have issued a reset of the sonoff pow and the counter went to 0 after boot start.

[image]

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-07-04:
This question comes up now and then and always receives the same answer: Invest some time in flash usage and you'll learn that saving changes every 5 minutes will lead your pow to heaven in no time.

See other closed issues about the same "problem".

Solution: let your Home Automation tool store values like Domoticz:

[image]

andreibanaru, 2018-07-04:
Theo,

The restart of the sonoff was planned. Had to work on a wire. But this might happen during power outages too.
I can overcome that reset to null value during a restart throught an optimized search in Splunk.
Thanks for the heads up on the flash storage.

I have a second question - Can the same thing be achieved via MQTT? Like with the LWT messages.
Sonoff to publish the values inside LWT and when it comes back online after a restart take the last value stored inside the brokers database.

What do you think?

arendst, 2018-07-04:
Might be possible. Interesting idea...

andreibanaru, 2018-07-05:
Yep. We lose the recording between the last retained value and power down.
In my case, I have TelePeriod = 30 so losing the power consumption for 30 seconds it's not that bad.

andrethomas, 2018-07-06:
I pick it up through mqtt on node-red and push the results into a database. My tele period is 60 seconds though... this way I can store usage data per device for a very long period and run all kinds of queries against the sql database.

The other reason I push it to a mysql database is to perform some scripting to get week on week, month on month, year on year statistics...

So its 10AM now... what did I use by this time yesterday? Running lower or higher than usual... do we allow geysers to switch on if we're already using more elec than the day before, or if the average weekly usage on geysers have exceeded 10kw/day... Nice things you can do once you have the data in sql... you're limited only by your imagination

andrethomas, 2018-07-06:
Ever so slightly different way of using the data, thats all.

[image]

andrethomas, 2018-07-06:
[image]

ascillato2, 2018-07-16:
Hi

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2