arendst/Tasmota #3129: Flash Pow R2 with Raspberry pi

turner-prize, 2018-07-04:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ ] Hardware: Raspberry pi 3

Hi folks. Has anyone managed to flash Tasmota to a POW r2 with a raspberry pi? I've soldered some pin headers on and followed the guide on the wiki but every time i run 

esptool.py --port /dev/ttyS0 erase_flash

i get the following:

esptool.py v2.4.1
Serial port /dev/ttyS0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Invalid head of packet (0x00)

Has anyone managed to successfully flash their POW r2 this way? or am i better getting an FTDI adaptor?

Frogmore42, 2018-07-04:
I have never tried flashing with a rpi, but I have flashed a lot of different things with a lot of different adapters.  While it can often feel like it is impossible, it is usually just a matter of making sure the thing getting flashed has proper power, that you have the serial lines hooked up correctly and that you have the thing you are trying to flash in the correct mode to accept flashing.

The error message indicates at least one of those is not correct for your attempt.  Try it wirhout the esp8266 attached and see if you get the same error message.
Try it with it attached, but not in bootloader mode to see if you get the same message.  Try a terminal emulator to see if you can see the serial output from the esp8266 to ensure you have the serial lines and power hooked up correctly.

Jason2866, 2018-07-05:
Flashing with a PI is complicated when you do it correct. Since the Pi use TTL Level (=5 Volt) you have to use a level shifter to 3.3 Volt! Not using can or will damage the ESP because it is specified for 3.3 Volt! 
The IOs are NOT 5 Volt tolerant.
So go get a USB seriall Adapter but NOT a FTDI one. The deliever not enough current on the 3.3V pin!
Buy a Adapter with a CH340 chip which can be switched between 3.3 Volt and 5 Volt.

Frogmore42, 2018-07-05:
Raspberry PI's have 3V3 GPIOs
http://www.mosaic-industries.com/embedded-systems/microcontroller-projects/raspberry-pi/gpio-pin-electrical-specifications

As far as the esp8266's tolerance, there is some debate.  3V3 is safe, but some believe 5V is okay too.  https://hackaday.io/page/2024-esp8266-is-5v-tolerant-after-all

But I agree with Jason, get a USB adapter, people have had better luck with the CH340 based ones.  The FTDI ones also work great if you do NOT use them to power the esp8266.

Jason2866, 2018-07-05:
@Frogmore42 
Thanks for the Info about the PI. I have seen so many "things" connected to the Pi with 5 Volt level....
So i suggest it is 5 Volt. Once again RTFM!
I use the Pi for OpenHab and 2 more with Volumio with a DAC shield, no DIY hardware connected.
I am a ESP and Atmega guy for "playing around" Pi is to expensive when somrthing goes wrong.

Frogmore42, 2018-07-05:
I agree that the Pi is too much when all you want to do is talk to a sensor or blink an LED.  I have a project for a connected camera, where the power of the Pi is probably necessary, but have not had enough time to get around to implementing it.

Also, since the logic levels for 3V3 and 5V are close and many inputs are 5V tolerant (at least for a little while) you can get away with bad designs frequently, but not always.

Frogmore42, 2018-07-05:
https://github.com/arendst/Sonoff-Tasmota/issues/3136 for someone who was able to make it work.

RuhanSA079, 2018-07-06:
Did you test your serial pins by tying them together (loopback) and opening a serial terminal, and see if there is any text printed/returned? Also check if your device is in the correct boot mode? Hold down the button while applying power to the device.

turner-prize, 2018-07-06:
I take it you mean tying the serial pins together on the pi? No I haven't tried that. would it just be 
~]$ screen /dev/ttyS0 to see any output?

Yeah I'm pretty sure it's in boot mode, I've done it so many times at this point! The only annoying thing is that since the light doesn't turn on there's no indication that it's actually in boot mode which is quite frustrating! Thanks for the tips, I'll try the pins tonight.

ascillato2, 2018-07-10:
Hi,

Have you managed to solve your issue?

turner-prize, 2018-07-10:
I haven't, but I've ordered a serial adaptor to flash my sonoff instead so I'll close this as the problem with the Pi isn't an issue for me any longer. Thank you all for your comments and help.

closed 2018-07-10 by turner-prize

mrgluek, 2018-12-03:
Got flashed with Raspberry Pi and using 5V instead of 3V3 port on Pi.
When you do that red (working) light on sonoff lighting a bit.
Tried 3-5 times and got success.

lalo-uy, 2018-12-03:
You wired PI 5V before Sonoff regulator ? Puting 5V on the 3,3V pin will
bourn the CPU.

mrgluek, 2018-12-03:
Pi's GPIO 5v pin to 3.3v on sonoff's VDD.

RuhanSA079, 2018-12-03:
The Sonoff pins is NOT 5V tolerant, as @lalo-uy said, it will burn the ESP8266 chip. In your case, it can become very hot, at the point it will either burn out the chip, or cause very serious damages. I burnt one ESP8266 chip that way, I did not pay attention to the FTDI chip voltage jumper. Nevertheless, it worked until the chip died.

mrgluek, 2018-12-03:
Okay, that's not the right way, but it worked, flashed 2 sonoff's pow r2 (which not responding to flashing with ftdi adapter), it's only takes 10-15 seconds, so may be it saved them from burning.

Frogmore42, 2018-12-03:
As it says in the wiki, the FTDI adapter (real one) is rated to NOT support the current at 3V3 that an esp8266 requires.  Many people are successful using it to flash, so it does work sometimes.

The esp8266 is also rated to not work at 5V.  I too have applied 5V to or more.  They all worked fine, until they didn't anymore.  Was it because I applied 5V, or for some other unknown reason.  We may never know for sure, but it seems very likely that was the reason.  If yours is working, great, use it and don't worry.  But, I would order a backup unit to have on hand for when it fails.

stuartornum, 2020-01-22:
Appreciate this is an old issue, so for reference - this is totally doable using a Raspberry Pi following this guide: https://github.com/arendst/Tasmota/wiki/Flash-Sonoff-using-Raspberry-Pi 

Just flashed three POW R2 using a RPi3 - note, I skipped the `esptool.py --port /dev/ttyS0 erase_flash` as this bricked two of my POW R2's previously.