arendst/Tasmota #3130: Magichome turns ons randomly

jongdesteven, 2018-07-05:
My magichome seems to turn on by itself, on random times.

My logging shows at that time:
```
05:42:08 MQT: Attempting connection...
05:42:13 MQT: Connected
05:42:14 MQT: home/keukeneiland/tele/LWT = Online (retained)
05:42:14 MQT: home/keukeneiland/cmnd/POWER = 
05:42:14 MQT: Subscribe to home/keukeneiland/cmnd/#
05:42:15 MQT: Subscribe to home/sonoffs/cmnd/#
05:42:16 RSL: Group 0, Index 1, Command POWER, Data OFF
05:42:16 MQT: home/keukeneiland/stat/RESULT = {"POWER":"OFF"}
05:42:16 MQT: home/keukeneiland/stat/POWER = OFF
05:42:16 MQT: Subscribe to cmnd/DVES_95724C/#
05:42:17 RSL: Group 0, Index 1, Command DIMMER, Data 55
05:42:17 MQT: home/keukeneiland/stat/RESULT = {"POWER":"ON","Dimmer":55,"Color":"3853","HSBColor":"0,0,55","Channel":[22,33],"CT":360}
05:42:17 RSL: Group 0, Index 1, Command CT, Data 352
05:42:17 MQT: home/keukeneiland/stat/RESULT = {"POWER":"ON","Dimmer":55,"Color":"3B50","HSBColor":"0,0,55","Channel":[23,31],"CT":352}
05:42:17 CFG: Saved to flash at F9, Count 961, Bytes 2560
05:46:07 MQT: home/keukeneiland/tele/STATE = {"Time":"2018-07-05T05:46:07","Uptime":"0T23:44:26","Vcc":3.169,"POWER":"ON","Dimmer":55,"Color":"3B50","HSBColor":"0,0,55","Channel":[23,31],"CT":352,"Fade":"OFF","Speed":20,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"De Jong Airport","RSSI":48,"APMac":"74:EA:3A:BD:0D:F4"}}
05:51:07 MQT: home/keukeneiland/tele/STATE = {"Time":"2018-07-05T05:51:07","Uptime":"0T23:49:26","Vcc":3.175,"POWER":"ON","Dimmer":55,"Color":"3B50","HSBColor":"0,0,55","Channel":[23,31],"CT":352,"Fade":"OFF","Speed":20,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"De Jong Airport","RSSI":48,"APMac":"74:EA:3A:BD:0D:F4"}}
05:56:15 MQT: home/keukeneiland/tele/STATE = {"Time":"2018-07-05T05:56:15","Uptime":"0T23:54:34","Vcc":3.178,"POWER":"ON","Dimmer":55,"Color":"3B50","HSBColor":"0,0,55","Channel":[23,31],"CT":352,"Fade":"OFF","Speed":20,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"De Jong Airport","RSSI":46,"APMac":"74:EA:3A:BD:0D:F4"}}
05:58:39 HTP: Main Menu
05:58:45 HTP: Console
```

- [ x] Searched the problem in issues and in the wiki
- [ x] Hardware used : Magichome with 5.14 firmware
- [ x] Provide the output of command ``status 0`` :
```
06:04:47 CMD: status 0
06:04:47 RSL: Group 0, Index 1, Command STATUS, Data 0
06:04:47 MQT: home/keukeneiland/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"keukeneiland","ButtonTopic":"0","Power":1,"PowerOnState":0,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
06:04:47 MQT: home/keukeneiland/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"External System","Uptime":"1T00:03:06","StartupUTC":"2018-07-04T05:01:41","Sleep":0,"BootCount":124,"SaveCount":961,"SaveAddress":"F9000"}}
06:04:47 MQT: home/keukeneiland/stat/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":7,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
06:04:47 MQT: home/keukeneiland/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["De Jong Airport","indebuurt2"],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
06:04:47 MQT: home/keukeneiland/stat/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
06:04:47 MQT: home/keukeneiland/stat/STATUS5 = {"StatusNET":{"Hostname":"keukeneiland-4684","IPAddress":"10.1.1.218","Gateway":"10.1.1.2","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.4","Mac":"60:01:94:95:72:4C","Webserver":2,"WifiConfig":2}}
06:04:47 MQT: home/keukeneiland/stat/STATUS6 = {"StatusMQT":{"MqttHost":"raspi","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_95724C","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
06:04:47 MQT: home/keukeneiland/stat/STATUS7 = {"StatusTIM":{"UTC":"Thu Jul 05 05:04:47 2018","Local":"Thu Jul 05 06:04:47 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:53","Sunset":"20:56"}}
06:04:47 MQT: home/keukeneiland/stat/STATUS10 = {"StatusSNS":{"Time":"2018-07-05T06:04:47"}}
06:04:47 MQT: home/keukeneiland/stat/STATUS11 = {"StatusSTS":{"Time":"2018-07-05T06:04:47","Uptime":"1T00:03:06","Vcc":3.178,"POWER":"ON","Dimmer":55,"Color":"3B50","HSBColor":"0,0,55","Channel":[23,31],"CT":352,"Fade":"OFF","Speed":20,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"***","RSSI":48,"APMac":"74:EA:**"}}}
06:06:24 MQT: home/keukeneiland/tele/STATE = {"Time":"2018-07-05T06:06:24","Uptime":"1T00:04:43","Vcc":3.170,"POWER":"ON","Dimmer":55,"Color":"3B50","HSBColor":"0,0,55","Channel":[23,31],"CT":352,"Fade":"OFF","Speed":20,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"De Jong Airport","RSSI":50,"APMac":"74:EA:3A:BD:0D:F4"}}
```

The Dimmer seems to be set to 55 randomly, and that makes the magichome turn on?

Jason2866, 2018-07-05:
It looks to me that your broker has stored retained messages. The device loose connection
and when online again the broker send this retained ones. So your device switches...
Do you use HA, many users has problems with that. This issue is HA related not TASMOTA.
You can search the issues here for that....

osemmler, 2018-07-05:
I have similar problem with Sonoff T1 2CH Module connected to home assistant broker. It is switching on and off randomly.

Here is my log:
>00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_CA8F25, GroupTopic sonoffs) Version 6.0.0c-2_3_0
00:00:00 WIF: Connecting to AP1 buldnet in mode 11N as sonoff-3877...
00:00:03 WIF: Connected
00:00:03 DNS: Initialized
00:00:03 HTP: Web server active on sonoff-3877.local with IP address 192.168.xxxxx
00:00:05 MQT: Attempting connection...
10:34:11 MQT: Connected
10:34:11 MQT: tele/sonoff/LWT = Online (retained)
10:34:11 MQT: cmnd/sonoff/POWER = 
10:34:11 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff T1 2CH","Version":"6.0.0c","FallbackTopic":"DVES_CA8F25","GroupTopic":"sonoffs"}
10:34:11 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-3877","IPAddress":"192.168.1.180"}
10:34:11 MQT: tele/sonoff/INFO3 = {"RestartReason":"Power on"}
10:34:13 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
10:34:13 MQT: stat/sonoff/POWER1 = ON
10:34:13 MQT: stat/sonoff/RESULT = {"POWER2":"ON"}
10:34:13 MQT: stat/sonoff/POWER2 = ON
10:34:14 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
10:34:14 MQT: stat/sonoff/POWER2 = OFF
10:34:20 MQT: tele/sonoff/STATE = {"Time":"2018-07-05T10:34:20","Uptime":"0T00:00:14","Vcc":3.220,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"buldnet","RSSI":64,"APMac":"74:D0:2B:58:F1:24"}}
10:34:30 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
10:34:30 MQT: stat/sonoff/POWER1 = OFF
10:35:10 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
10:35:10 MQT: stat/sonoff/POWER1 = ON
10:35:23 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
10:35:23 MQT: stat/sonoff/POWER1 = OFF
10:35:57 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
10:35:57 MQT: stat/sonoff/POWER1 = ON
10:37:36 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
10:37:36 MQT: stat/sonoff/POWER1 = OFF
10:38:17 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
10:38:17 MQT: stat/sonoff/POWER1 = ON
10:39:20 MQT: tele/sonoff/STATE = {"Time":"2018-07-05T10:39:20","Uptime":"0T00:05:14","Vcc":3.222,"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"buldnet","RSSI":48,"APMac":"74:D0:2B:58:F1:24"}}
10:39:55 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
10:39:55 MQT: stat/sonoff/POWER1 = OFF

reloxx13, 2018-07-05:
this is tasmota git for the tasmota firmware and not for HA or "how to configre my ha correctly".

your logs says clearly your mqtt broker is sending the signals.

jongdesteven, 2018-07-05:
@Jason2866 Thanks for the hint. 
I have purged the retained commands in my mosquitto and it seems to be resolved so far.

closed 2018-07-05 by jongdesteven

osemmler, 2018-07-06:
@jongdesteven Could you please help me how to solve it? I dissconnect HA from power and my Sonoff T1 2CH Module is still randomly switch on and off...
What do you mean with?
> I have purged the retained commands in my mosquitto
Could this help me even if there is problem without HA (sonoff is not connected to any broker).

jongdesteven, 2018-07-06:
@osemmler 

1) Make sure in HA, you have set:
`retain: false`
2) purge the retained messages with: 
`mosquitto_pub -t cmnd/sonoff/POWER -n -r`

osemmler, 2018-07-07:
@jongdesteven @Jason2866 HA is powered off (no broker connected to sonoff) and still switching on and off randomly. There are ghost button presses in log console:

> 10:37:52 HTP: Main Menu
10:37:53 HTP: Console
10:37:56 APP: Button1 multi-press 1
10:37:59 MQT: Attempting connection...
10:38:06 MQT: Connect failed to 192.168.1.51:1883, rc -2. Retry in 10 sec
10:38:06 RSL: stat/sonoff/RESULT = {"POWER1":"ON"}
10:38:06 RSL: stat/sonoff/POWER1 = ON
10:38:06 CFG: Saved to flash at F5, Count 63, Bytes 2560
10:38:14 WIF: Checking connection...
10:38:14 WIF: Connected

osemmler, 2018-07-08:
Just figured out, that problem was with unstable power supply.

qingz2004, 2018-08-05:
I got the same problem.
Are you sure it is caused by unstable power supply?
What does "APP: Button1 multi-press 1" mean? What is APP here?