arendst/Tasmota #3132: webserver password incorrect

charinduedirisuriya, 2018-07-05:
I flashed tasmota with hardcoded WEB_USERNAME and WEB_PASSWORD in the user_config_override.h and in the user_config.h (also i uncommented #define USE_CONFIG_OVERRIDE) file. How ever i cannot seem to login to the web ui with the new credentials. Before this i had another version of tasmota flashed around 2 months ago and now i flashed the lastest version with these settings. As a work around what i did was i reset the device with MQTT because the device was connected to WiFi and MQTT before, and now it works.
Why didn't it work previously, did i forget to include something in the user_config file or is this just a bug?

Frogmore42, 2018-07-05:
https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#config-problems-can-cause-boot-loops-items-set-in-user_config-to-not-appear-etc

ascillato2, 2018-07-10:
Hi,

Your question seems that has been addressed. Closing this issue. Please, reopen it, if needed. Thanks.

closed 2018-07-10 by ascillato2