arendst/Tasmota #3133: Define a GPIO for SDM120 SDM630 to control serial direction of RS485

Karl-opec, 2018-07-05:
Could the modules for reading SDM120 and SDM630 extended to provide RS485 direction
control?

hahmann, 2018-08-13:
Hi Karl, not sure it will work for you but as I'm using only one RS485 device, I was able to address it using an automatic enable Rx/Tx using the diagram below I got from https://electronics.stackexchange.com/questions/65881/any-good-way-to-automatically-enable-an-rs-485-transmitter-in-hardware

[image]

I'm using it right now with no issues.

Karl-opec, 2018-08-14:
Thanks for your Hint. I know these type of solutions for generating direction control. But i would like to see a software controlled direction pin (GPIO) because the software knows best in a RS485 bus which direction the data should flow.

ascillato2, 2018-09-04:
Hi,

Why is this necessary? Tasmota already support SDM120

Karl-opec, 2018-09-04:
The SDM120/630 use RS485 signals. RS485 is in this case a half duplex communication. Therefore the direction of communication flow needs to switch the drivers for the wires. (SEND vs. RECEIVE). This signal is needed to do this (GPIO=1 ==> SEND, GPIO=0 ==> RECEIVE).

This is basicly needed for good RS485 Half Duplex Communication not only for SDM120 /SDM630

ascillato, 2018-09-04:
Ok, I understand, but now the SDM120 is working in Tasmota. So, what feature/reading is being added if having this extra pin?

@gtortone what do you think about this?

Karl-opec, 2018-09-05:
If you look at a typical RS485 driver chip you have PIN for TX /RX A/B for the RS485 wires and the direction control /RE and DE. the additional GPIO signal would control the direction.

[image]

if the esp device is basicly in listening mode to switch the driver to receive , if it is going to send the rs485 devices it switch the direction control GPIO to send and starts transmitting after end of transmission it is switching this gpio back to receive. Normaly Uarts have RTS/CTS signals to do this.

the drawing in comment 2 shows a simple electronic help circuit to simulate this direction control. But to have a Signal from the bus master for that is better because only the master knows really when to send.

ascillato2, 2018-09-06:
Hi,

There is an open issue about RS485 implementation at https://github.com/arendst/Sonoff-Tasmota/issues/2803

Let's add all this information to that proposal and continue the thread there. Thanks

closed 2018-09-06 by ascillato2

AdrielGH, 2019-05-01:
Hi @ascillato2,

Is it possible to directly connect a SDM630 to Sonoff/Tasmota? Or should I have a conversor RS485 to TTL?

Tks

gtortone, 2019-05-01:
Hi @AdrielGH,
you need a RS485-TTL adapter like explained here:
https://github.com/jarcikw/SDM120_Homeassistant/wiki/SDM120-&-nodemcu-&-RS485-wiring

Regards,
Gennaro

AdrielGH, 2019-05-02:
Thank you Gennaro!!

AdrielGH, 2019-05-28:
Hi @ascillato2,

I bought RS485-TTL and connected it according to:
https://github.com/jarcikw/SDM120_Homeassistant/wiki/SDM120-&-nodemcu-&-RS485-wiring

It is working with a Sonoff 4CH Pro and Tasmota 6.5.0 I getting all information from SDM630... it is amazing.

But after connect SDM630(RS485-TTL) to Sonoff 4CH Pro it started to restart showing the follow message:  

MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}

It restarts many many times per hour... I suspected that it could be a power supply issue becouse I was using 3.3V from sonoff as VCC to  power the RS485-TTL , so I connected an external 5V power supply to RS485-TTL, my Sonoff is receiving informatio from SDM630, but unfortunately still restarting with message "Software/System restart".

Do you have any idea about what I should try to solve this software restart issue?

Thanks,

Adriel

ascillato, 2019-05-29:
Hi,

When restarting with "Software/System restart" is because Tasmota have received a restart command or a command that requires a restart. Please, check for retained messages in your mqtt broker.

If that don't solve your issue, please, address this to the Tasmota support chat. Thanks

AdrielGH, 2019-05-29:
Hi @ascillato2,

I disabled mqtt (SetOption3 off) in order to not receive any mqtt comand from external brokers.

Even with mqtt disabled it still restating "RestartReason":"Software/System restart".

Thanks anyway for the tip.

Saludos desde Brasil